Interbotix/interbotix_ros_manipulators #82: [Question]: Robot works fine but the ros2 control list_controllers command doesn't return the loaded controllers

AbishekNP, 2023-05-25:
### Question

I'm executing the "ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=vx300s" command along with a custom python script using the ROS-python API and the robot executes the trajectory correctly. But, when I execute the ros2 control list_controllers command to see the running controllers, it throws the "Could not contact service /controller_manager/list_controllers" error. I don't understand how the bot is able to perform the action but ROS returns that it couldn't find the controllers. Kindly help me with this issue.

### Robot Model

vx300s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-05-25:
The Python-ROS 2 API does not use controllers, rather it uses a custom IK and trajectory solver separate from ros2_control. Learn more in the [documentation](https://docs.trossenrobotics.com/interbotix_xsarms_docs/python_ros_interface.html) and [source code](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/rolling/interbotix_xs_toolbox/interbotix_xs_modules/interbotix_xs_modules/xs_robot/arm.py).

closed 2023-05-25 by LSinterbotix

AbishekNP, 2023-05-25:
Thanks a lot for the clarification!

AbishekNP, 2023-05-25:
Just to clarify. Doesn't the Python-ROS1 API use ros_control as well?

LSinterbotix, 2023-05-25:
I'm not sure what you're asking exactly but the only interbotix_ros_xsarms packages that use controllers from ros2_control are the [xsarm_ros_control](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros2_packages/ros_control.html) and [xsarm_moveit](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros2_packages/moveit_motion_planning_configuration.html) packages. xsarm_ros_control configures and launches the joint_trajectory_controller arm and gripper [controllers](https://github.com/Interbotix/interbotix_ros_manipulators/tree/rolling/interbotix_ros_xsarms/interbotix_xsarm_ros_control/config/controllers) and xsarm_moveit then uses those launched controllers to follow trajectories generated by MoveIt.

Hopefully this answers your question.

AbishekNP, 2023-05-25:
My goal is to modify the existing controllers used in the motors. I launched the **xsarm_control** package and ran a custom Python script developed using the Python-ROS2 API and tried listing the controllers currently being used using the **"ros2 control list_controllers"** command but got the above-mentioned output. From my understanding of your previous reply, only when launching the **xsarm_ros_control** and/or **xsarm_moveit** packages, the "**ros2 control list_controllers**" command will return the ROS controllers in use(Correct me if I'm wrong). In that case, is there any other way for me to modify the controllers being used if I plan on using the **xsarm_control** package and not **xsarm_ros_control** or **xsarm_moveit**?

LSinterbotix, 2023-05-25:
You are correct, we only use ros2_controllers in the xsarm_ros_control and xsarm_moveit packages.

If you're asking about the package that interacts directly with the arm's motors, you should look at the [xs_driver](https://github.com/Interbotix/interbotix_xs_driver/tree/c9ce74e54369197b6e07f5608d86882e4affe283), the [xs_sdk](https://github.com/Interbotix/interbotix_ros_core/tree/humble/interbotix_ros_xseries/interbotix_xs_sdk) (the xs_driver's ROS 2 wrapper), and its dependencies. You can see that the xs_sdk is launched by the xsarm_control package below:
https://github.com/Interbotix/interbotix_ros_manipulators/blob/3ba0e5e55edabb4a116396e9af96e9228409ac88/interbotix_ros_xsarms/interbotix_xsarm_control/launch/xsarm_control.launch.py#L88-L104

If you can provide a bit more information on what you're trying to modify, I might be able to give you more specific guidance.

AbishekNP, 2023-05-25:
I believe I've got a lot more clarity now. I have a final question though...can we use the ROS2-Python API with the interbotix_xsarm_ros_control package?

LSinterbotix, 2023-05-25:
The API does not have integrations with any of the controllers configured and launched in xsarm_ros_control, though this is something we'd like to have the ability to do in the future. You can still send commands to the xs_sdk using the API (i.e. calling services or commanding joint positions) since it's launched underneath the ros_controllers, it just doesn't currently have the ability to command the JointTrajectoryController.

AbishekNP, 2023-05-25:
So, the xsarm_ros_control and xsarm_control packages can interface with a custom developed ROS node that publishes joint trajectory values, but the Python-API is only meant to be used with xsarm_control and the xs_sdk packages. Did I get it right?

LSinterbotix, 2023-05-25:
Yes, that is correct.

AbishekNP, 2023-05-25:
Awesome! Thanks a lot for your time and guidance.