Interbotix/interbotix_ros_manipulators #86: [Question]: "./xsarm_amd64_install.sh: line 234: /home/vaishnavi/interbotix_ws/devel/setup.bash: No such file or directory" I faced this error during the installation of ros1 noetic interface and cannot work further as it shows that devel is missing in further stages. how to resolve these issues?

Ansh090303, 2023-06-09:
### Question

"./xsarm_amd64_install.sh: line 234: /home/vaishnavi/interbotix_ws/devel/setup.bash: No such file or directory" i faced this error while installing the ros1 noetic interface and cannot work further as it shows devel is missing in further step. how to resolve this issue?

### Robot Model

rx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2023-06-12:
Please share the console output of the installation script when you see this error.

Ansh090303, 2023-06-23:
> Do you have any further questions on this? Otherwise, this issue will be
> closed.

I solved the issue, thanks a lot. I don’t have any further questions

closed 2023-06-23 by LSinterbotix