Interbotix/interbotix_ros_manipulators #87: [Question]: Control the force exerted by end-effector

AbishekNP, 2023-06-13:
### Question

My application requires to move the end-effector to a particular point and exert a certain amount of force on that point (Eg: Press a button). I'm using the ROS2-Python API to program the joint trajectory. Is there some way to control the end-effector force using the ROS2-Python API? 

Thank You

### Robot Model

VX300s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-06-13:
Force control via the Python-ROS/2 API is not supported and is not currently planned.

closed 2023-06-13 by LSinterbotix

LSinterbotix, 2023-06-14:
> control the end-effector force

By end-effector I'm assuming that you don't mean controlling the gripper fingers and more something like whole-arm [impedance control](https://en.wikipedia.org/wiki/Impedance_control)?

This is not possible without modifications to the underlying xs_sdk.

There is a current-based position [operating mode](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#operating-mode11) available on the motors. You should be able to use this mode to limit current in each joint, though this mode does not respect joint position limits.

AbishekNP, 2023-06-14:
Not the gripper fingers. I've replaced the gripper fingers with a custom end-effector for my application. I'd require the end-effector  to move to a particular position on the target surface and apply a certain amount of force on it.