Interbotix/interbotix_ros_manipulators #88: [Question]: There are two questions

Frozen-Dream, 2023-06-15:
### Question

#### 1. Correlation packet deletion:

When I execute ```ros2 pkg list | grep interbotix```, it shows:

```
interbotix_common_modules
interbotix_common_sim
interbotix_common_toolbox
interbotix_moveit_interface
interbotix_moveit_interface_msgs
interbotix_ros_xsarms
interbotix_ros_xsarms_examples
interbotix_ros_xseries
interbotix_rpi_modules
interbotix_rpi_msgs
interbotix_rpi_toolbox
interbotix_tf_tools
interbotix_xs_driver
interbotix_xs_modules
interbotix_xs_msgs
interbotix_xs_ros_control
interbotix_xs_rviz
interbotix_xs_sdk
interbotix_xs_toolbox
interbotix_xsarm_control
interbotix_xsarm_descriptions
interbotix_xsarm_joy
interbotix_xsarm_moveit
interbotix_xsarm_moveit_interface
interbotix_xsarm_ros_control
interbotix_xsarm_sim**
```

I don't have any package related to perception.

#### 2. Some errors reported:

When I execute ```ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=wx250s hardware_type:=actual```, it has some errors:

```
robot@WP:~/interbotix_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=wx250s hardware_type:=actual
[INFO] [launch]: All log files can be found below /home/robot/.ros/log/2023-06-15-21-10-48-705324-WP-15187
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [15207]
[INFO] [rviz2-2]: process started with pid [15209]
[INFO] [ros2_control_node-3]: process started with pid [15211]
[INFO] [spawner-4]: process started with pid [15213]
[INFO] [spawner-5]: process started with pid [15215]
[INFO] [xs_sdk-6]: process started with pid [15217]
[INFO] [robot_state_publisher-7]: process started with pid [15219]
[xs_sdk-6] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.3'.
[xs_sdk-6] [INFO] Using logging level 'INFO'.
[xs_sdk-6] [INFO] Loaded mode configs from '/home/robot/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/modes.yaml'.
[xs_sdk-6] [INFO] Loaded motor configs from '/home/robot/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/wx250s.yaml'.
[move_group-1] [WARN] [1686834649.903149935] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1686834649.903192785] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1686834649.905659539] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00233334 seconds
[move_group-1] [INFO] [1686834649.905696496] [moveit_robot_model.robot_model]: Loading robot model 'wx250s'...
[move_group-1] [INFO] [1686834649.905711789] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] Link wx250s/base_link had 1 children
[move_group-1] Link wx250s/shoulder_link had 1 children
[move_group-1] Link wx250s/upper_arm_link had 1 children
[move_group-1] Link wx250s/upper_forearm_link had 1 children
[move_group-1] Link wx250s/lower_forearm_link had 1 children
[move_group-1] Link wx250s/wrist_link had 1 children
[move_group-1] Link wx250s/gripper_link had 1 children
[move_group-1] Link wx250s/ee_arm_link had 2 children
[move_group-1] Link wx250s/gripper_prop_link had 0 children
[... 74 lines cut ...]
[ros2_control_node-3] [INFO] [1686834651.696291762] [wx250s.controller_manager]:   controller_manager/test_controller
[ros2_control_node-3] [INFO] [1686834651.696317316] [wx250s.controller_manager]:   controller_manager/test_controller_failed_init
[ros2_control_node-3] [INFO] [1686834651.696341949] [wx250s.controller_manager]:   controller_manager/test_controller_with_interfaces
[ros2_control_node-3] [INFO] [1686834651.698135045] [wx250s.controller_manager]: Loading controller 'gripper_controller'
[ros2_control_node-3] [ERROR] [1686834651.698226738] [wx250s.controller_manager]: Loader for controller 'gripper_controller' not found.
[ros2_control_node-3] [INFO] [1686834651.698249081] [wx250s.controller_manager]: Available classes:
[ros2_control_node-3] [INFO] [1686834651.698268086] [wx250s.controller_manager]:   controller_manager/test_controller
[ros2_control_node-3] [INFO] [1686834651.698283206] [wx250s.controller_manager]:   controller_manager/test_controller_failed_init
[ros2_control_node-3] [INFO] [1686834651.698297403] [wx250s.controller_manager]:   controller_manager/test_controller_with_interfaces
[xs_sdk-6] [INFO] [1686834651.737579777] [interbotix_xs_sdk.xs_sdk]: InterbotixRobotXS is up!
[ERROR] [spawner-4]: process has died [pid 15213, exit code 1, cmd '/opt/ros/galactic/lib/controller_manager/spawner -c /wx250s/controller_manager arm_controller --ros-args -r __node:=arm_controller_spawner -r __ns:=/wx250s'].
[ERROR] [spawner-5]: process has died [pid 15215, exit code 1, cmd '/opt/ros/galactic/lib/controller_manager/spawner -c /wx250s/controller_manager gripper_controller --ros-args -r __node:=gripper_controller_spawner -r __ns:=/wx250s'].
```

### Robot Model

wx250s

### Operating System

Ubuntu 20.04

### ROS Version

ROS 2 Galactic

LSinterbotix, 2023-06-15:
1. The perception packages are not built by default on Raspberry Pis. Please delete the IGNORE files in the perception directory, run [rosdep](http://wiki.ros.org/rosdep#Install_dependency_of_all_packages_in_the_workspace), and rebuild your workspace.
2. Check that you have `ros-galactic-ompl` and `ros-galactic-moveit-planners-ompl` installed.

Frozen-Dream, 2023-06-16:
Following your suggestion, I made modifications. After that, the Motion Planning column in the visualization interface still displays an error, which is what is displayed on the command line:
```
robot@WP:~/interbotix_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=wx250s hardware_type:=actual
[INFO] [launch]: All log files can be found below /home/robot/.ros/log/2023-06-16-18-35-47-093073-WP-27442
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [27462]
[INFO] [rviz2-2]: process started with pid [27464]
[INFO] [ros2_control_node-3]: process started with pid [27466]
[INFO] [spawner-4]: process started with pid [27468]
[INFO] [spawner-5]: process started with pid [27470]
[INFO] [xs_sdk-6]: process started with pid [27472]
[INFO] [robot_state_publisher-7]: process started with pid [27474]
[xs_sdk-6] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.3'.
[xs_sdk-6] [INFO] Using logging level 'INFO'.
[xs_sdk-6] [INFO] Loaded mode configs from '/home/robot/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/modes.yaml'.
[xs_sdk-6] [INFO] Loaded motor configs from '/home/robot/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/wx250s.yaml'.
[move_group-1] [WARN] [1686911748.315388874] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1686911748.315434254] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1686911748.318005797] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00242501 seconds
[move_group-1] [INFO] [1686911748.318039275] [moveit_robot_model.robot_model]: Loading robot model 'wx250s'...
[move_group-1] [INFO] [1686911748.318048670] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] Link wx250s/base_link had 1 children
[move_group-1] Link wx250s/shoulder_link had 1 children
[move_group-1] Link wx250s/upper_arm_link had 1 children
[move_group-1] Link wx250s/upper_forearm_link had 1 children
[move_group-1] Link wx250s/lower_forearm_link had 1 children
[move_group-1] Link wx250s/wrist_link had 1 children
[move_group-1] Link wx250s/gripper_link had 1 children
[move_group-1] Link wx250s/ee_arm_link had 2 children
[move_group-1] Link wx250s/gripper_prop_link had 0 children
[move_group-1] Link wx250s/gripper_bar_link had 1 children
[move_group-1] Link wx250s/fingers_link had 3 children
[move_group-1] Link wx250s/ee_gripper_link had 0 children
[move_group-1] Link wx250s/left_finger_link had 0 children
[move_group-1] Link wx250s/right_finger_link had 0 children
[move_group-1] [INFO] [1686911748.376956152] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1686911748.377153177] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/wx250s/joint_states' for joint states
[move_group-1] [INFO] [1686911748.377452738] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/wx250s/joint_states'
[move_group-1] [INFO] [1686911748.377797395] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1686911748.377809901] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1686911748.378456663] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1686911748.378468646] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1686911748.378721495] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1686911748.379023290] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1686911748.379316673] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1686911748.379334139] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1686911748.379345381] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1686911748.382991493] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1686911748.403002882] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1686911748.405561442] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1686911748.405581872] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1686911748.405586404] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[... cut ...]
[ros2_control_node-3] [INFO] [1686911750.275771027] [wx250s.arm_controller]: Controller state will be published at 50.00 Hz.
[ros2_control_node-3] [INFO] [1686911750.276153051] [wx250s.arm_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-4] [INFO] [1686911750.469585141] [wx250s.arm_controller_spawner]: Configured and started arm_controller
[INFO] [spawner-4]: process has finished cleanly [pid 27468]
[spawner-5] [INFO] [1686911750.669623867] [wx250s.gripper_controller_spawner]: Configured and started gripper_controller
[INFO] [spawner-5]: process has finished cleanly [pid 27470]

```

LSinterbotix, 2023-06-23:
> the Motion Planning column in the visualization interface still displays an error

Please provide a screenshot of this error.

Frozen-Dream, 2023-06-26:
Problem solved, thank you

closed 2023-06-26 by LSinterbotix