Interbotix/interbotix_ros_manipulators #92: [Bug]: Computer cannot connect to U2D2

TriNguyen2002, 2023-07-05:
### What happened?

i did the instruction on the page,until the command ls /dev | grep ttyDXL didn't return anything. I also have tried the dmesg and it returned:
```
[ 3372.508332] usb 1-3: device not accepting address 46, error -71
[ 3372.508486] usb usb1-port3: unable to enumerate USB device
[ 3495.736662] usb usb1-port2: Cannot enable. Maybe the USB cable is bad?
[ 3496.636647] usb usb1-port2: Cannot enable. Maybe the USB cable is bad?
[ 3496.636774] usb usb1-port2: attempt power cycle
```
I have tried different ports, different cables and still the same results. People online suggested that it might have been a hardware problem or some drivers is corrupted

### Robot Model

rx150

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Humble

### Steps To Reproduce

i have followed this instruction: [https://www.makeuseof.com/tag/fix-usb-device-port-linux/](url)

LSinterbotix, 2023-07-05:
Two things:

1. Are you able to detect the U2D2 on other computers?
2. See if [this troubleshooting tip](https://docs.trossenrobotics.com/interbotix_xsarms_docs/troubleshooting.html#u2d2-not-recognized-no-symlink-created) helps.

TriNguyen2002, 2023-07-06:
The U2D2 was fine on other computers, and the lsusb didn't return anything when i plug in the robot.
I'm thinking my Ubuntu has some problems since when i type sudo update, the terminal spit out some weird errors about the initrd. I will reinstall Ubuntu tomorrow and try again.

TriNguyen2002, 2023-07-07:
After the reinstallation, my computer still won't recognize the U2D2.

TriNguyen2002, 2023-07-07:
i don't know if the udev rules file are correct, this is the file :
`# Place this file in /etc/udev/rules.d/
# Then reload udev by typing 'udevadm control --reload-rules && udevadm trigger'
# Sets up rules to give permanent names to devices

# Allow serial devices to be read by anyone
KERNEL=="ttyUSB*", MODE:="0666"
KERNEL=="ttyACM*", MODE:="0666"
KERNEL=="js*", MODE:="0666"
KERNEL=="video*", MODE:="0666"

# OpenCM9.04C board
SUBSYSTEM=="tty", ATTRS{idVendor}=="fff1", ATTRS{idProduct}=="ff48", ENV{ID_MM_DEVICE_IGNORE}="1", SYMLINK+="ttyDXL"

# U2D2 board (also sets latency timer to 1ms for faster communication)
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6014", ENV{ID_MM_DEVICE_IGNORE}="1", ATTR{device/latency_timer}="1", SYMLINK+="ttyDXL"

# RPLidar
SUBSYSTEM=="tty", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60", SYMLINK+="rplidar"

# Kobuki base
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="kobuki*", ATTR{device/latency_timer}="1", MODE:="0666", GROUP:="dialout", SYMLINK+="kobuki"

# LifeCam Cinema
SUBSYSTEM=="video4linux", ATTRS{idVendor}=="045e", ATTRS{idProduct}=="0812", ATTR{index}=="0", SYMLINK+="lifecam"
`

LSinterbotix, 2023-07-10:
That udev rules file is correct.

Run `ls /dev | grep ttyUSB` before and after plugging in the U2D2. Does a new device appear?

TriNguyen2002, 2023-07-14:
`lsusb ` didn't return anything from FTDI, and i've tried to plug and unplug it, didn't show up at all. 
Since i'm using dual boot on the laptop (Acer Aspire A715-41G), I have also booted into window to check and found out that Window 10 cannot find any connection to U2D2, even after i installed the VCP driver for Window. Are there any problems about the compatibility of Acer's laptops ?

LSinterbotix, 2023-07-14:
My own development computer is an Acer Nitro so I don't think it has anything to do with that.

Email trsupport at trossenrobotics dot com and link this issue.

closed 2023-09-26 by LSinterbotix