Interbotix/interbotix_ros_manipulators #93: [Bug]: AttributeError: type object 'type' has no attribute '_TYPE_SUPPORT'

sekisushai, 2023-07-05:
### What happened?

Hello,
I'm trying to run the `bartender.py` demo on simulated robot and I run into the following error:
```
AttributeError: type object 'type' has no attribute '_TYPE_SUPPORT'
```

### Robot Model

wx250

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Humble

### Steps To Reproduce

1. Launch the robot in simulation mode:
```
ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=wx250 use_sim:=true
```
2. Launch `bartender.py`
```
python bartender.py
```

### Relevant log output

```shell
File "/opt/ros/humble/local/lib/python3.10/dist-packages/trajectory_msgs/msg/_joint_trajectory.py", line 46, in __import_type_support__
    if Header.__class__._TYPE_SUPPORT is None:

AttributeError: type object 'type' has no attribute '_TYPE_SUPPORT'
```
```

LSinterbotix, 2023-07-05:
Please provide the full traceback.

Searching the error message lead me to [this issue](https://github.com/ros2/ros2/issues/451#issuecomment-360735965) which states that it might be an issue with your `PYTHONPATH` environment variable. Check if that is the case for you as well.

sekisushai, 2023-07-06:
It appends in spyder IDE only. Otherwise it is working using `python bartender.py`. I've checked the `PYTHONPATH` and they are identical in spyder and in the terminal..

closed 2023-07-06 by sekisushai