Interbotix/interbotix_ros_manipulators #94: [Question]: Using the Moveit2 API

AbishekNP, 2023-07-11:
### Question

Hello...
I'm trying to use the MoveIt2 API to send joint angle values to the arm, but the arm remains static in it's home pose.
This is how I've modified the mvoeit_interface.cpp file.
KIndly help me out if I'm missing something.
Thank You

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  rclcpp::NodeOptions node_options;
  node_options.automatically_declare_parameters_from_overrides(true);
  auto node = std::make_shared<rclcpp::Node>("interbotix_moveit_node", node_options);
  auto moveit_interface = std::make_shared<interbotix::InterbotixMoveItInterface>(node);
  auto executor = std::make_shared<rclcpp::executors::MultiThreadedExecutor>();
  executor->add_node(node);
  std::vector<double> joint_positions = {1.503, -0.34, 1.076, 0.0, 0.37, 1.486, 0.00, 0.02, -0.02};
  
  bool success = moveit_interface->moveit_plan_joint_positions(joint_positions);
  moveit_interface->moveit_execute_plan();

  executor->spin();
  rclcpp::shutdown();
  return 0;
}

### Robot Model

vx300s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

AbishekNP, 2023-07-12:
Hey. I'm still facing the same issue. I'd really appreciate your help.

LSinterbotix, 2023-07-12:
You are using the interface incorrectly. You should use the `moveit_plan` service it provides to plan using an external node or script. See the [moveit_interface_gui](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/342d9f2600b52183038dcc22a5f30b0c67046ee1/interbotix_common_toolbox/interbotix_moveit_interface/scripts/moveit_interface_gui) for an example on how to send these requests.

Additionally, you are sending requests before spinning the executor, preventing anything from being executed. This would cause the non-responsive behavior you're seeing.

AbishekNP, 2023-07-12:
Hey. Thanks a lot for the example code. I'll try it out and let you know. 

Also, I was looking at the MoveitPlan.srv file and found that the end-effector position is planned with respect to the ee_arm_link and world frame. Shouldn't it be with respect to the ee_gripper_link and world frame?

LSinterbotix, 2023-07-12:
The documentation in the service definition is wrong. Planning is done based on the robot's [SRDF's end_effector tag](https://github.com/Interbotix/interbotix_ros_manipulators/blob/cd6679d4f09dd7e8d47fd985d377341446cc72df/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/srdf/vx300s.srdf.xacro#L73). You can see the reference frame and end effector link printed out [on the interface's startup](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/45cc26d6e2c82cc1cd2b649eee0ff7676f63b3d0/interbotix_common_toolbox/interbotix_moveit_interface/src/moveit_interface_obj.cpp#L18-L22).

Thank you for catching this.

AbishekNP, 2023-07-12:
Thanks a lot for the clarification. I'm still working on understanding the API. The source code of  moveit_interface_gui, where you made calls to the API, is in python. Do you have a similar or different file where the API calls are made in C++?

Thanks again!

LSinterbotix, 2023-07-24:
We do not have such a demo for C++.

closed 2023-08-14 by LSinterbotix