Interbotix/interbotix_ros_manipulators #96: [Bug]:

AbishekNP, 2023-07-18:
### What happened?

I'm trying to install the Interbotix packages on a Raspberry Pi 4b. 
I've downloaded the right version of OS (Ubuntu 22.04.02 LTS 64 bit for RPi4/400 models with 2Gb+) and etched it to the SD card.

I tried executing these commands:
sudo apt install curl
curl 'https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/main/interbotix_ros_xsarms/install/rpi4/xsarm_rpi4_install.sh' > xsarm_rpi4_install.sh
chmod +x xsarm_rpi4_install.sh
./xsarm_rpi4_install.sh -d humble

The xsarm_rpi4_install.sh executable is getting downloaded, but when I try executing it, I get the following error:
[ERROR] Something went wrong
[ERROR] Interbotix Remote installation Failed! 

I currently don't have any version of ROS running on the Pi as it was stated in the documentation that the script will automatically download them.

### Robot Model

vx300s

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Humble

LSinterbotix, 2023-07-24:
The rpi4 installation script was not updated for ROS 2 Humble but should now work properly. Please redownload the script any try again.

LSinterbotix, 2023-08-10:
Were you able to get ROS 2 Humble installed on your Pi using the installation script? If so, feel free to close this issue.

closed 2023-08-21 by LSinterbotix