Interbotix/interbotix_ros_manipulators #107: [Question]: Tuning the PID controller

AbishekNP, 2023-08-25:
### Question

Hello

I've customized the arm's gripper and added 350-400g of payload to it. I'm trying to produce a straight line trajectory between two points in the joint space but the arm struggles to move straightly and a curve is observed in it's motion. I'm not extending the arm beyond 50% of it's length as mentioned in the documentation. I haven't made any modifications to the existing PID gains.

Is the current payload too much for the arm or do the PID gains need tuning for the new payload?

Thank you

### Robot Model

vx300s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-09-06:
The motor gains can be tuned for increased accuracy with additional payload. See [this issue](https://github.com/Interbotix/interbotix_ros_manipulators/issues/24) for details. In addition to the steps mentioned in the issue, you can also set the gains in the motor config file. For example, to change the [`Position_P_Gain`](https://emanual.robotis.com/docs/en/dxl/x/xm540-w270/#position-p-gain) on the shoulder joint too 1000 instead of the default 800, you would add the lines below to the shoulder motors configuration block:

```diff
  shoulder:
    ID: 2
    Baud_Rate: 3
    Return_Delay_Time: 0
    Drive_Mode: 1
    Velocity_Limit: 131
    Min_Position_Limit: 841
    Max_Position_Limit: 2867
+   Position_P_Gain: 1000
    Secondary_ID: 255

  shoulder_shadow:
    ID: 3
    Baud_Rate: 3
    Return_Delay_Time: 0
    Drive_Mode: 0
    Velocity_Limit: 131
    Min_Position_Limit: 841
    Max_Position_Limit: 2867
+   Position_P_Gain: 1000
    Secondary_ID: 2
```

AbishekNP, 2023-09-11:
Hello
Thanks for the help. I noticed that the default values for Position I gain and Position D gain are set as 0 by default. Are they meant to be left at their default values or could those gains be modified for a better performance as well?

LSinterbotix, 2023-09-14:
Feel free to tune the gains as necessary. It is best to tune these values slowly and in small increments as you may experience overshooting, overtorquing, or unstable behavior as the gains are set higher.

closed 2023-10-03 by LSinterbotix