Interbotix/interbotix_ros_manipulators #110: [Bug]: wx250s wrist_link inverted rviz

vyeevani, 2023-08-30:
### What happened?

Following getting started guide and the rviz is showing the flipped position of the wrist_link.

### Robot Model

Wx250s

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Rolling

### Steps To Reproduce

Launch rviz
Launch control.launch
Turn off all motor torque

### Additional Info

This is a new robot that I’m setting up for the first time. Possible a hardware issue with the motor installed the wrong way?

LSinterbotix, 2023-08-30:
1. Provide a photo of the real robot and its virtual representation in RViz where the flipped angle can clearly be seen.
2. Have the URDF or motor config file been modified in any way?
3. What launch command(s) did you use?
4. Provide the output of the launch command(s).

vyeevani, 2023-08-31:
$ ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=wx250s
[INFO] [launch]: All log files can be found below /home/vineeth/.ros/log/2023-08-30-10-02-10-350763-cyberdyne-60541
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [xs_sdk-1]: process started with pid [60574]
[INFO] [robot_state_publisher-2]: process started with pid [60576]
[INFO] [rviz2-3]: process started with pid [60578]
[xs_sdk-1] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.3'.
[xs_sdk-1] [INFO] Using logging level 'INFO'.
[xs_sdk-1] [INFO] Loaded mode configs from '/home/vineeth/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/modes.yaml'.
[xs_sdk-1] [INFO] Loaded motor configs from '/home/vineeth/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/wx250s.yaml'.
[xs_sdk-1] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  8, Model: 'XL430-W250', Joint Name: 'wrist_rotate'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  7, Model: 'XM430-W350', Joint Name: 'wrist_angle'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  5, Model: 'XM430-W350', Joint Name: 'elbow_shadow'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  4, Model: 'XM430-W350', Joint Name: 'elbow'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  6, Model: 'XM430-W350', Joint Name: 'forearm_roll'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XM430-W350', Joint Name: 'shoulder'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  9, Model: 'XL430-W250', Joint Name: 'gripper'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XM430-W350', Joint Name: 'shoulder_shadow'.
[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XM430-W350', Joint Name: 'waist'.
[xs_sdk-1] [WARN] Writing startup register values to EEPROM. This only needs to be done once on a robot if using a default motor config file, or after a motor config file has been modified. Can set `write_eeprom_on_startup` to false from now on.
[xs_sdk-1] [INFO] The operating mode for the 'arm' group was changed to 'position' with profile type 'time'.
[xs_sdk-1] [INFO] The operating mode for the 'gripper' joint was changed to 'pwm' with profile type 'velocity'.
[xs_sdk-1] [INFO] Interbotix X-Series Driver is up!
[xs_sdk-1] [INFO] [1693414932.390944937] [interbotix_xs_sdk.xs_sdk]: InterbotixRobotXS is up!
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[xs_sdk-1] [INFO] [1693491932.627127000] [rclcpp]: signal_handler(signum=2)
[INFO] [rviz2-3]: process has finished cleanly [pid 60578]
[INFO] [xs_sdk-1]: process has finished cleanly [pid 60574]
[ERROR] [robot_state_publisher-2]: process has died [pid 60576, exit code -11, cmd '/opt/ros/humble/lib/robot_state_publisher/robot_state_publisher --ros-args -r __ns:=/wx250s --params-file /tmp/launch_params_8q184co7'].

$ ros2 service call /wx250s/torque_enable interbotix_xs_msgs/srv/TorqueEnable "{cmd_type: 'group', name: 'all', enable: false}"
requester: making request: interbotix_xs_msgs.srv.TorqueEnable_Request(cmd_type='group', name='all', enable=False)

response:
interbotix_xs_msgs.srv.TorqueEnable_Response()

vyeevani, 2023-08-31:
https://github.com/Interbotix/interbotix_ros_manipulators/assets/30946190/234aab15-70a5-470f-9847-b9afbafa93df

vyeevani, 2023-08-31:
I'd like to know if this is an issue with rviz or something about the hardware itself.

LSinterbotix, 2023-09-05:
I suspect this is an issue with the hardware. Please contact our [technical support](https://www.trossenrobotics.com/contact.aspx) for assistance and reference this issue in your ticket. We will leave this open until your issue is solved.

vyeevani, 2023-09-09:
Swapping the drive mode of the wrist_angle motor to 0 from the default of 1 fixed the issue. I'm not sure if this something specific to my hardware or something out of sync between the default wx250s config but I'm leaving this issue open for anyone else who may face this.

vyeevani, 2023-09-11:
Turns out that the forearm_roll motor was installed 180 degrees the wrong way which led to all the problems that were observed.

LSinterbotix, 2023-09-11:
You can find the default drive mode for each motor in the arm in [its motor config file](https://github.com/Interbotix/interbotix_ros_manipulators/blob/rolling/interbotix_ros_xsarms/interbotix_xsarm_control/config/wx250s.yaml). Feel free to close this issue if you feel like your problem has been solved.

closed 2023-10-03 by LSinterbotix