Interbotix/interbotix_ros_manipulators #113: [Question]: I use pix100 robot arm for my class. We have 4 of those and one of them gives "cannot find dynamixel error"

madibabaiasl, 2023-08-31:
### Question

I use Pincherx 100 robot arm to teach robotics in my class and 3 out of 4 are working fine. For one of them when connecting the physical robot arm, I got the following errors:

[xs_sdk-1] [INFO] 	Found DYNAMIXEL ID:  4, Model: 'XL430-W250', Joint Name: 'wrist_angle'.
[xs_sdk-1] [ERROR] 	Can't find DYNAMIXEL ID:  3, Joint Name: 'elbow':
[xs_sdk-1] 		  '[TxRxResult] There is no status packet!'
[xs_sdk-1] [ERROR] 	Can't find DYNAMIXEL ID:  2, Joint Name: 'shoulder':
[xs_sdk-1] 		  '[TxRxResult] There is no status packet!'
[xs_sdk-1] [ERROR] 	Can't find DYNAMIXEL ID:  5, Joint Name: 'gripper':
[xs_sdk-1] 		  '[TxRxResult] There is no status packet!'
[xs_sdk-1] [ERROR] 	Can't find DYNAMIXEL ID:  1, Joint Name: 'waist':
[xs_sdk-1] 		  '[TxRxResult] There is no status packet!'
[xs_sdk-1] [FATAL] Failed to find all motors. Shutting down...

I tried to troubleshoot according to troubleshooting page of the website and using the Dynamixel wizard, but the wizard only shows the ID 4 servo. What is the solution? 

### Robot Model

px100

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-09-06:
A few things:

- Search all baudrates when running the DW tool.
- Unplug and replug all 3-pin connectors between motors.
- You may attempt a [firmware recovery](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-recovery) on the problem motors.
  - If this is the case, make sure to have only a single motors attached to the U2D2 during the process.
  - All motors in the px100 are XL430-W250. See the [default servo configuration guide](https://docs.trossenrobotics.com/interbotix_xsarms_docs/specifications/px100.html#default-servo-configurations) for more details.

LSinterbotix, 2023-10-09:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2023-10-09 by LSinterbotix