Interbotix/interbotix_ros_manipulators #114: [Bug]: Not able to calibrate the plane of RViz simulator

MurshidKc010, 2023-09-07:
### What happened?

So we launched the actual arm through RViz but the problem is simulation is not same as that of the physical arm, for example in the sleep position it is showing like this in the RVIz, ideally in the simulator also it should come to the sleep position. Could you please help in this?
[image]

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 1 Noetic

### Additional Info

So in your documentation for wx 200, it shows the series for servo motors like this
[image]
But while connecting through the Dynamixel wizard, it shows like this
[image]
Ideally what should we select the series for the servo motors, I think like that this might cause this problem

LSinterbotix, 2023-09-08:
Could you check that all of the motors on your arm match the model in the Default Servo Configurations table? The motors' models are displayed on their faces.

MurshidKc010, 2023-09-11:
Thank you for the response, We changed the motor series for wrist angle to XL430-W250. After changing to this series it is not even detecting in the Dynamixel Wizard so we reverted the changes. But the model displayed on their face is XM430-W350 only. What do we have to do here? It would be really helpful if you could provide a response really quick

LSinterbotix, 2023-09-11:
Try running a [firmware recovery](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-recovery) on the problem motor, setting its model to the one displayed on its face. Make sure only the one motor is plugged into the U2D2 when running through this process.

MurshidKc010, 2023-09-12:
We already did this also, in Your WX-200 documentation shows that the motor series of wrist angle, wrist rotate, and gripper as XL430-W250 series but in the face it is showing XM430-W350 series for these motors, so we maintained these motors as XM430-W350 series. But still, the simulation is not happening properly, when the arm moves in one direction in the simulator it is moving in the opposite direction

LSinterbotix, 2023-09-12:
Please provide a picture of the problem motor and the way it is mounted.

MurshidKc010, 2023-09-12:
This is the rest pose of the arm in actual and in the simulator. Can you let us know why are they both different? Also we have figured out by changing the rpy and xyz values in the wx-200 URDF file, the calibration can be fixed. I hope we are correct here. Can you help us with the correct values of each motors and links so that we can fix this?
[image]
[image]
[image]
[image]

MurshidKc010, 2023-09-13:
Any reply please?
Waiting for the response fro m yourside @LSinterbotix

LSinterbotix, 2023-09-13:
Please contact our [technical support](https://www.trossenrobotics.com/contact.aspx) for assistance and reference this issue in your ticket.

LSinterbotix, 2023-09-14:
The robot isn't actually "playing" chess, it's just moving the pieces around. The chessboard and chess pieces were used as a way to demonstrate the capabilities of our perception pipeline.

One of our customers did actually manage to get their robot to play against them. You can read more about it in [this discussion](https://github.com/orgs/Interbotix/discussions/6).

closed 2023-10-03 by LSinterbotix