Interbotix/interbotix_ros_manipulators #115: [Bug]: wx250s -> parameter 'robot_description_planning.joint_limits.left_finger.max_velocity' has invalid type: expected [double] got [integer]

TenguBoss, 2023-09-07:
### What happened?

Aloha InterbotiX Team. I believe I have found a bug in the joint limits files. Please see the steps below.

### Robot Model

wx250s

### Operating System

Other (Describe in "Additional Info")

### ROS Distro

ROS 2 Humble

### Steps To Reproduce

FIND BUG
--------
When running the moveit demo, without the robot, with

`~$ source ~/interbotix_ws/install/setup.bash`
`~$ ros2 launch interbotix_xsarm_moveit_interface xsarm_moveit_interface.launch.py robot_model:=wx250s dof:=6 hardware_type:=fake use_cpp_interface:=true`

The following error shows up in the output:

...
[move_group-3] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[move_group-3]   what():  parameter 'robot_description_planning.joint_limits.left_finger.max_velocity' has invalid type: expected [double] got [integer]
...

and the rviz interface is stuck in 

[moveit_interface-1] [WARN] [1694111650.493413867] [moveit_ros.planning_scene_interface.planning_scene_interface]: service '/get_planning_scene' not advertised yet. Continue waiting...

The interface with sliders doesn't show up, the MotionPlanning item in RViz
is orange and it has a a warning sign (!) in front.

FIX BUG
-------

Find all the yaml files whose name contains "250s" and "joint" that contain the
string 'left_finger' with the following command:

`~$ find -name "*250s*joint*.yaml" -exec grep -lR 'left_finger' {} \;`

./interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/joint_limits/wx250s_joint_limits.yaml
./interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/joint_limits/mobile_wx250s_joint_limits.yaml
./interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/joint_limits/wx250s_joint_limits.yaml
./interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/joint_limits/mobile_wx250s_joint_limits.yaml

Confirm that each of the first three files has the following entries:

```
left_finger:
     has_velocity_limits: true
     max_velocity: 1
     has_acceleration_limits: true
     max_acceleration: 0.01
right_finger:
     has_velocity_limits: true
     max_velocity: 1
     has_acceleration_limits: true
     max_acceleration: 0.01
```
 
where the max_velocity value has been entered as an integer for both the 
left_finger and right_finger joints.

Modified the max_velocity lines in each of the first three files to

    max_velocity: 1.0

### Relevant log output

```shell
See above
```

### Additional Info

Operating system: WSL2 under Windows 11

Linux cookiethumper 5.15.90.4-microsoft-standard-WSL2 #1 SMP Tue Jul 18 21:28:32 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

LSinterbotix, 2023-09-09:
The `max_velocity` problem should be fixed in 777c2711f8c85523af096871ff3cc2c0dd78c61f. Let me know if those changes work for you.

TenguBoss, 2023-09-10:
Aloha LSinterbotix. The changes are working, thank you for the prompt fix.

closed 2023-09-11 by LSinterbotix