Interbotix/interbotix_ros_manipulators #117: [Question]: ModuleNotFoundError: No module named 'interbotix_xs_modules.arm'

edenespin, 2023-09-08:
### Question

Hi,

I am working with the rx200 robot arm. However, I cannot run any python demos without getting the error "ModuleNotFoundError: No module named 'interbotix_xs_modules.arm'". I have the arm.py file in the following path: ~/interbotix/src/interbotix_ros_toolboxes/interbotix_xs_toolbox/interbotix_xs_modules/interbotix_xs_modules/xs_robot. The python demos are in this path: ~/interbotix/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/python_demos. How can I fix this error to run the bartender.py program?

Thank you.

### Robot Model

rx200

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

### Additional Info

Duplicated from Interbotix Discussion.

LSinterbotix, 2023-09-08:
The path you provided to the Python Demos is not the correct path for the ROS 2 version of our packages. Based on that, I believe something went wrong during the install. Did you use one of our [standard installation scripts](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/software_setup.html#amd64-architecture) to download and install the Interbotix packages?

LSinterbotix, 2023-09-25:
Do you have any further questions on this? Otherwise, this issue will be closed.

edenespin, 2023-09-26:
So I had to uninstall and reinstall the software package, and this solved the error message about the missing arm.py. However, I now have a new problem where I connect to Rviz using $ ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=wx200 (from the QuickStart Guide), but the arm isn't showing up. Rviz starts and displays its grid, but there's no arm in the center. How can I fix this?

LSinterbotix, 2023-09-26:
> I now have a new problem

Please close this issue and open a new one. When you do, make sure to provide detailed information like the contents of your launch terminal, screenshots, etc.

closed 2023-09-26 by edenespin