Interbotix/interbotix_ros_manipulators #118: [Question]: Help with running a database for handling persistent scenses and states

TenguBoss, 2023-09-11:
### Question

I am new to ROS and I am trying to figure out a way to run a storage plugin for "persistent scenes and states" with the MoveIt demo from Trossen Robotics/InterbotiX that is launched with

`$ ros2 launch interbotix_xsarm_moveit_interface xsarm_moveit_interface.launch.py robot_model:=wx250s dof:=6 hardware_type:=fake use_cpp_interface:=false`

I am trying to follow the example here [https://moveit.picknik.ai/humble/doc/examples/persistent_scenes_and_states/persistent_scenes_and_states.html](url) that talks about a file called `move_group.launch.py` I haven't been able to find this file on my computer. The closest seems to be `~/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_moveit_interface/launch/xsarm_moveit_interface.launch.py`

Morevoer, the instructions in the URL above mention a "Move Group config" and I have tried to add the `run_move_group_node` to `~/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/launch/xsarm_moveit.launch.py` with a few package names that I have guessed. My guesses at the package names are incorrect so I am getting a segmentation fault when I try to run the demo mentioned above.

Is there any documentation available on how to change the scripts and configuration files in the InterbotiX MoveIt demo or packages, in general, to set up a storage plugin for persistent scenes and states?

Thank you.
BU

### Robot Model

wx250s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2023-09-14:
We created our launch files before the MoveIt2 Setup Assistant was released and, as such, do not have the standard MoveIt package structure.

If you'd like to take a crack at implementing the database configuration yourself, you can look around the xsarm_moveit.launch.py launch file to find all launched nodes and some relevant configuration.

move_group node:
https://github.com/Interbotix/interbotix_ros_manipulators/blob/777c2711f8c85523af096871ff3cc2c0dd78c61f/interbotix_ros_xsarms/interbotix_xsarm_moveit/launch/xsarm_moveit.launch.py#L188-L214

MoveIt-configuration RViz node:

configuration:
https://github.com/Interbotix/interbotix_ros_manipulators/blob/777c2711f8c85523af096871ff3cc2c0dd78c61f/interbotix_ros_xsarms/interbotix_xsarm_moveit/launch/xsarm_moveit.launch.py#L114-L171

TenguBoss, 2023-09-15:
@LSinterbotix thank you for letting me know. I have started building an application from scratch but I'll try them today and let you know how it works.

LSinterbotix, 2023-09-18:
Great to hear that fixed it. Feel free to close this issue if you don't have any further questions.

closed 2023-09-25 by TenguBoss