Interbotix/interbotix_ros_manipulators #122: [Question]: Arm not displaying on Rviz?

edenespin, 2023-09-26:
### Question

Hi,
I have a problem where I connect to Rviz using $ ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=rx200 (from the QuickStart Guide), but the arm isn't showing up. Rviz starts and displays its grid, but there's no arm in the center. How can I fix this?
[image]
[image]

### Robot Model

rx200

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

### Additional Info

I had previously uninstalled the startup software code then reinstalled it. The arm displayed on Rviz in the beginning but recently stopped. I tried installing the software on other computers and to no avail as well. I think it is more of an arm problem than it is the software.

edenespin, 2023-09-27:
I have an update with my problem:
I did receive the warning about the register values being written on EEPROM, but this was solved by adding "load_configs:=false" at the end of the launch command. However, Rviz still has the Global Status warning that "No tf data. Actual error: Fixed Frame [base_link] does not exist." How can I fix this?

LSinterbotix, 2023-09-27:
In another terminal, run the command below to check if you can discover ROS 2 traffic.

```
ros2 node list
```

Please also provide the RMW that you're using by checking the output of the command:

```
printenv RMW_IMPLEMENTATION
```

edenespin, 2023-09-27:
this is what happens

LSinterbotix, 2023-09-27:
Have you previously configured an RMW? Did your network conditions change? Sometimes restarting the ROS 2 daemon (or even your computer) can solve the issue.

Are you able to see the nodes if you run the command without the daemon?

```
ros2 node list --no-daemon
```

Can you see any topics?

```
ros2 topic list
ros2 topic list --no-daemon
```

Try installing `rmw_cyclonedds_cpp` and setting your RMW to that instead of `rmw_fastrtps_cpp` and try again.

edenespin, 2023-09-28:
Update:
After installing rmw-cyclonedds (https://github.com/ros2/rmw_cyclonedds), I have to explicitly type RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 launch interbotix_xsarm_control ... and it loads the model into Rviz correctly.

However, I am now receiving a "Waiting for service to become available" notice after trying to run the provided demos (such as bartender.py). This message lasts indefinitely and nothing happens on either Rviz nor on the arm model itself. Is this because I switched RMWs, or is this a separate issue?

LSinterbotix, 2023-09-29:
You can append the line below to your `~/.bashrc` file so that each new terminal knows what implementation it should use.

```bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

edenespin, 2023-10-02:
Ok, I will do that. But does switching RMWs affect how I should run the demos? After having run the launch command, I open another terminal and $cd into the Interbotix_ws directory with the demos, then try to run bartender.py with python3. I keep getting "Waiting for service to become available..." in this terminal window.

edenespin, 2023-10-05:
It turns out I just needed to clear ROS2 off my computer and reinstall it. This way returns to using the default RMW. Thanks for your help!

closed 2023-10-05 by edenespin