Interbotix/interbotix_ros_manipulators #123: [Question]: I need a stl file of one of the parts of the arm since it is broken.

MurshidKc010, 2023-09-28:
### Question

The part that is under the gripper broke while we were working with the arm; we need a STL file for that so that we can 3D print it, or this part is alone available so that we can buy it.

[image]
[image]

Also the gripper and cartridge also we 3D printed, but we didn't get any screws for joining them, could you provide any link or dimension of them so that we can buy it

[image]

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2023-09-28:
The swing arm component is laser cut 1.5mm delrin. Here is the DXF: [SwingArm-430-D-1.5mm.zip](https://github.com/Interbotix/interbotix_ros_manipulators/files/12752433/SwingArm-430-D-1.5mm.zip)

Could you take a clear picture of where the missing hardware should go for the gripper and cartridge?

MurshidKc010, 2023-09-29:
Can I have an option to buy this swing arm from your side, if yes please provide the link from where we can buy it.

LSinterbotix, 2023-09-29:
Yes, you can email our team at the Technical Support email on [our contact page](https://www.trossenrobotics.com/contact.aspx).

manishvel8, 2023-10-03:
> The swing arm component is laser cut 1.5mm delrin. Here is the DXF: [SwingArm-430-D-1.5mm.zip](https://github.com/Interbotix/interbotix_ros_manipulators/files/12752433/SwingArm-430-D-1.5mm.zip)
> @LSinterbotix Can you please provide the stl file for the swing arm component, which is laser cut from 1.5mm delrin? After converting DXF to STL files, we are facing issues while 3D printing. So, can you please send the STL file?

LSinterbotix, 2023-10-03:
The original component is laser cut from delrin so the performance may not be the same. You can contact our support at the link I shared previously to get a real replacement.

Here is the DXF converted to an STL: [SwingArm-430-D-1.5mm-stl.zip](https://github.com/Interbotix/interbotix_ros_manipulators/files/12796488/SwingArm-430-D-1.5mm-stl.zip)

closed 2023-10-27 by LSinterbotix