Interbotix/interbotix_ros_manipulators #130: [Bug]: velocity limits in ros2 iron

m-elwin, 2023-10-16:
### What happened?

I know that ros2 iron is not yet officially supported, but currently using moveit with the `hardware_type:=fake` does not work.
It seems like changing  due to `has_velocity_limits: false` in the joint_limit config files (i.e., the files in `interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_moveit/config/joint_limits`
setting these velocity limits allows the plan to execute.

### Robot Model

px100

### Operating System

Ubuntu 22.04

### ROS Distro

Other (Describe in "Additional Info")

### Steps To Reproduce

`ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=px100 hardware_type:=fake`
Then plan a path with moveit (e.g., in Rviz). That will work. But executing the plan will fail.

### Relevant log output

```shell
[move_group-1] [INFO] [1697493699.598969875] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: /px100/arm_controller started execution
[move_group-1] [WARN] [1697493699.599017431] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request rejected
[move_group-1] [ERROR] [1697493699.599059060] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal was rejected by server
[move_group-1] [ERROR] [1697493699.599130972] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller /px100/arm_controller
[move_group-1] [INFO] [1697493699.599143929] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1697493699.599285112] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
```

### Additional Info

Working with ros2 Iron on Ubuntu 22.04

LSinterbotix, 2023-10-17:
Thanks for reporting this. I was able to reproduce the issue and your recommended fix worked for me. Let me know if 81ea963 fixes the issue.

m-elwin, 2023-10-18:
Works for me thank you!

commit 81ea963f (2023-10-17): [xsarm_moveit] Add required joint velocity limits

Fixes #130.
closed 2023-10-18 by m-elwin