Interbotix/interbotix_ros_manipulators #132: [Question]:

manishvel8, 2023-10-24:
### Question

Hi Team,
We are planning to play chess with our Widowx 200 robotic arm. It's like a human vs. robot arm; for that, first we are doing a virtual simulation. How to do a virtual simulation using our arms in chess Are there any open-source projects for virtual simulations available, or are there any reference links available? Also, we need URDF file to play chess with our robot arm in virtual simulation.

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

manishvel8, 2023-10-27:
Team, 
Any suggestions for this request? @LSinterbotix

LSinterbotix, 2023-10-27:
You can find URDFs/xacros of our arms in the interbotix_xsarm_descripitons package. I don't personally know of any chess-playing simulators out there; your search will be as good as mine.

closed 2023-10-27 by LSinterbotix

mohanr2609, 2023-10-30:
shall we use rviz or gazebo interface for simulation chess playing