Interbotix/interbotix_ros_manipulators #136: Is Interbotix wx200 will able to play full chess game with human or not

kethan637, 2023-11-06:
### Question

1. The main objective of the project playing chess with an interbotix wx200 arm is that possible.
2. Suppose that is possible how to do it can you give me an idea?or suggestion
3. One more thing those in-built packages sufficient to play chess or not
4. If chess can be played can you send the reference to do 

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2023-11-06:
1. The arm should be able to do whatever you program it to do. We've had a customer modify their robot's end effector and create a program that allows it to play chess. See [this discussion](https://github.com/orgs/Interbotix/discussions/6) for more details.
2. This is a pretty open-ended question and there are a lot of ways to accomplish it. You will definitely need to create your own over-arching program.
3. Depends on how you create your program. I can see a few of the packages being useful like the xsarm_moveit package and the xsarm_perception package.
4. See the link in 1.

kethan637, 2023-11-07:
In that discussion we were able to see videos but there was no code for reference  soo can you send any of the GitHub code for reference for the code part.
Any idea for chessboard coordinate detection

kethan637, 2023-11-07:
And how can i find the coordinate of wx200 arm is there code or some were how to find that

LSinterbotix, 2023-11-07:
> can you send any of the GitHub code for reference for the code part

We do not have this code since this was done by a customer.

> Any idea for chessboard coordinate detection

A quick Google search for "chessboard ros" provides a few results. You could use AprilTags to calibrate the position of some part of the chessboard relative to the camera or create a known fixed transform between the base and the chessboard and include that in whatever program you use to control the arm.

> how can i find the coordinate of wx200 arm

You can find transforms between the arm's linkages and joints in [its URDF](https://github.com/Interbotix/interbotix_ros_manipulators/blob/main/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/wx200.urdf.xacro).

closed 2023-11-22 by LSinterbotix