Interbotix/interbotix_ros_manipulators #137: [Question]:  How to know the current PID gains of the motor?

AbishekNP, 2023-11-08:
### Question

Hello.
I know how to adjust the PID gains using the set/motor_pid_gains service. But I'm not sure how to know what the current (default) PID gains are. 
May I please know if there's a way to know that? 
If this is not possible, may I please know the gain values you tuned on the arm when you shipped it?

Thank you.

### Robot Model

vx300s

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2023-11-08:
You can find default motor gains on the Robotis e-Manual page for each servo. For example, the XM540-W270 gains are in the RAM table [here](https://emanual.robotis.com/docs/en/dxl/x/xm540-w270/#control-table-of-ram-area). Note that because these gains are stored in RAM, they are reset whenever the servo is powered off.

AbishekNP, 2023-11-09:
Thank you for your response. Will the changes be permanent if I modify it directly in the modes.yaml file instead of doing a service call each time?

LSinterbotix, 2023-11-09:
The ideal place to change these would be in the vx300s.yaml motor configuration file under each motor.

closed 2023-11-22 by LSinterbotix