Interbotix/interbotix_ros_manipulators #138: [Question]: Profile Velocity adjustment

AbishekNP, 2023-11-08:
### Question

I have two questions regarding adjusting the profile velocity.

Ques 1: Is it possible to adjust the profile velocity using a ROS service call in between a cartesian trajectory? 
For eg: Move to Goal A -> ROS service call to lower the profile velocity -> Move to Goal B -> ROS service call to increase the profile velocity -> Move to Goal C.

Is this possible or can't the velocity be adjusted once the robot has started a trajectory?

Ques 2: If it's possible to adjust the velocity, will it be possible to include a ROS service call to the set_motor_register service when using the MoveIt API?

Thank You

### Robot Model

vx300s

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

AbishekNP, 2023-11-10:
Hello. Could you kindly help me with this doubt?

LSinterbotix, 2023-11-14:
I can't see any reason why you couldn't do this. For example, in the Interboitx Python-ROS API, we allow users to specify certain timing parameters for overall movements and individual movements. See the [timing parameters section](https://docs.trossenrobotics.com/interbotix_xsarms_docs/python_ros_interface.html#timing-parameters) of the Python API docs and the set_ee_cartesian_trajectory [parameters](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/0c739cdab1dbab03d79e752b43fa3db14d5bb15e/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/arm.py#L254-L257)

You can do something similar to what we do with timing parameter service call requests in the [`set_trajectory_time`](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/0c739cdab1dbab03d79e752b43fa3db14d5bb15e/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/arm.py#L81-L87) method of the arm module.

closed 2023-12-11 by LSinterbotix