Interbotix/interbotix_ros_manipulators #140: About joints related

kethan637, 2023-11-17:
### Question

Actually, my question is when I ran the script with x=0.5 and y=0.16and z=0.2 it went correctly if we increase the x value it is  more than0.5 it doesn't go and when I use the pitch and roll option in the code is going up to x=0.35 only not more than that it is saying nopose in theta list variable so fix this any solution are there if there can you help us
->One more thing can we change the default values of the joints of wx200 arm if possible can you how to do so I can change and run this script

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2023-11-17:
Please provide the script you're using.

The wx200 arm has only 5 degrees of freedom which significantly limits its dextrous workspace. You will need to organize the robot's workspace in such a way that it is not impacted by this limitation.

The joint limits are there so the robot does not collide with itself.

kethan637, 2023-11-18:
from interbotix_sdk.robot_manipulation import InterbotixRobot
from interbotix_descriptions import interbotix_mr_descriptions as mrd
# This script makes the end-effector go to a specific pose by defining the
pose components
#
# To get started, open a terminal and type 'roslaunch interbotix_sdk
arm_run.launch robot_name:=wx250 use_time_based_profile:=true
gripper_operating_mode:=pwm'
# Then change to this directory and type 'python ee_pose_components.py'
def main():
arm = InterbotixRobot(robot_name="wx200", mrd=mrd)
arm.go_to_home_pose()
arm.set_ee_pose_components(x=0.2, y=0.1, z=0.2, roll=1.0, pitch=1.5)
arm.go_to_home_pose()
arm.go_to_sleep_pose()
if __name__=='__main__':
main()

if i use x more than 0.35 we are getting error is

.py
Robot Name: wx200
Robot Model: wx200
Initialized InterbotixRobotXSCore!

Arm Group Name: arm
Moving Time: 2.00 seconds
Acceleration Time: 0.30 seconds
Drive Mode: Time-Based-Profile
Initialized InterbotixArmXSInterface!

Gripper Name: gripper
Gripper Pressure: 50%
Initialized InterbotixGripperXSInterface!

[WARN] [1700210143.100437]: Would exceed position limits on joint 1.
[WARN] [1700210143.101849]: Limits are [-1.884956, 1.972222], value was
3.904000.
[WARN] [1700210143.126493]: Would exceed position limits on joint 2.
[WARN] [1700210143.128382]: Limits are [-1.884956, 1.623156], value was
-3.538000.
[WARN] [1700210143.154413]: Would exceed position limits on joint 2.
[WARN] [1700210143.156014]: Limits are [-1.884956, 1.623156], value was
-3.538000.
[WARN] [1700210143.157386]: No valid pose could be found. Returned
theta_list variable may be nonsense.
[WARN] [1700210144.336128]: No valid pose could be found. Returned
theta_list variable may be nonsense.
[WARN] [1700210150.437152]: Would exceed position limits on joint 1.
[WARN] [1700210150.438361]: Limits are [-1.884956, 1.972222], value was
-2.502000.
[WARN] [1700210150.467069]: Would exceed position limits on joint 2.
[WARN] [1700210150.468304]: Limits are [-1.884956, 1.623156], value was
-3.587000.
[WARN] [1700210150.494803]: Would exceed position limits on joint 2.
[WARN] [1700210150.495991]: Limits are [-1.884956, 1.623156], value was
-3.587000.
[WARN] [1700210150.496767]: No valid pose could be found. Returned
theta_list variable may be nonsense.

Can you say how to fix this issue and can you say how to change the default
values soo please help me out

kethan637, 2023-11-20:
can you reply for the above issue

closed 2023-12-11 by LSinterbotix