Interbotix/interbotix_ros_manipulators #142: Description.launch file

kethan637, 2023-11-29:
### Question

is possible to change control.launch file from the description .launch file for the changes of min and max value of gripper.
->If this possible please help how to change the that joint values of gripper.
->Major question is there any joints values for the gripper rotating mortar.
` <joint name="gripper" type="continuous">
    <axis xyz="1 0 0"/>
    <limit effort="1" velocity="${pi}"/>
    <origin rpy="0 0 0" xyz="0.0055 0 0"/>
    <parent link="$(arg robot_name)/ee_arm_link"/>
    <child link="$(arg robot_name)/gripper_prop_link"/>
    <dynamics friction="0.1"/>
  </joint>

  <transmission name="trans_gripper">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="gripper">
      <hardwareInterface>hardware_interface/EffortJointInterface</hardwareInterface>
    </joint>
    <actuator name="gripper_motor">
      <mechanicalReduction>1</mechanicalReduction>
    </actuator>
  </transmission>

  <link name="$(arg robot_name)/gripper_prop_link">
    <visual>
      <origin rpy="0 0 ${pi/2}" xyz="-0.0685 0 0"/>
      <geometry>
        <mesh filename="package://interbotix_xsarm_descriptions/meshes/wx200_meshes/wx200_7_gripper_prop.stl" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="interbotix_black"/>
    </visual>
    <collision>
      <origin rpy="0 0 ${pi/2}" xyz="-0.0685 0 0"/>
      <geometry>
        <mesh filename="package://interbotix_xsarm_descriptions/meshes/wx200_meshes/wx200_7_gripper_prop.stl" scale="0.001 0.001 0.001"/>
      </geometry>
    </collision>
    <inertial>
      <origin rpy="0 0 ${pi/2}" xyz="0.0008460000 -0.0000016817 0.0000420000"/>
      <mass value="0.00434" />
      <inertia ixx="0.0000005923" iyy="0.0000011156" izz="0.0000005743" ixy="0.0000000000" ixz="0.0000003195" iyz="-0.0000000004" />
    </inertial>
  </link>`

is this a correct code of that

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2023-11-29:
It is not currently possible to change these values in that way. As mentioned in [other issues](https://github.com/Interbotix/interbotix_ros_manipulators/issues/140#issuecomment-1823003091), it is not advised to change any joint limits.

The minimum joint limit may be fine to **increase** the [lower limit](https://github.com/Interbotix/interbotix_ros_manipulators/blob/341878954d92c7049d74eb8df2653f8012ea9dd1/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/wx200.urdf.xacro#L393C27-L393C40) of the left_finger joint and **decrease** the [upper limit](https://github.com/Interbotix/interbotix_ros_manipulators/blob/341878954d92c7049d74eb8df2653f8012ea9dd1/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/wx200.urdf.xacro#L433C42-L433C56) of the right_finger joint. Be cautious and do this at your own risk.

kethan637, 2023-11-30:
After changing values how to use that description.launch file where to save
and where to keep the file can you say please

kethan637, 2023-11-30:
After changing the values tell me the exact process to work with those changes values how to launch and process those values help me with steps of the process

LSinterbotix, 2023-11-30:
You don't need to do anything beyond changing the values in the [file I pointed out](https://github.com/Interbotix/interbotix_ros_manipulators/blob/341878954d92c7049d74eb8df2653f8012ea9dd1/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf/wx200.urdf.xacro). Once the limits are set in the URDF, the Interbotix Python-ROS API will use those new values as the minimum gripper limits i.e. they won't close past the new width you specified.

closed 2023-12-18 by LSinterbotix