Interbotix/interbotix_ros_manipulators #143: [Question]: Velocity control over vx300s using ROS

WangYixuan12, 2023-11-30:
### Question

As shown in the doc, we could control a robot using the joint position. I am wondering whether we could control the robot using joint velocity or torque. If we can, what is API for them? Thanks!

### Robot Model

vx300s

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2023-11-30:
You can send velocity (rad/s) commands to the motors when they are in the [velocity operating mode](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/overview/xs_msgs.html#interbotix-xs-msgs-operatingmodes-ros1). We don't have an API for this method of control exactly, but you can still use the [`InterbotixRobotXSCore`](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/0c739cdab1dbab03d79e752b43fa3db14d5bb15e/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/core.py#L15) class's [`robot_write_commands()`](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/0c739cdab1dbab03d79e752b43fa3db14d5bb15e/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/core.py#L131) and [`robot_write_joint_command()`](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/0c739cdab1dbab03d79e752b43fa3db14d5bb15e/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/core.py#L138C9-L138C34) methods.

LSinterbotix, 2023-12-11:
Do you have any further questions on this? Otherwise, this issue will be closed.

WangYixuan12, 2023-12-11:
Nope. Thanks for answering the question!

closed 2023-12-11 by WangYixuan12