Interbotix/interbotix_ros_manipulators #144: MuJoCo Menagerie Model -- URDF Request

kevinzakka, 2023-12-04:
### Question

I'd love to have a vx300 in [MuJoCo Menagerie](https://github.com/google-deepmind/mujoco_menagerie) and was wondering if you could provide me with a URDF since xacro refuses to work on my machine.

### Robot Model

vx300

### Operating System

Other (Describe in "Additional Info")

### ROS Version

Other (Describe in "Additional Info")

kevinzakka, 2023-12-04:
Also would love to have the widowx 250 :)

LSinterbotix, 2023-12-04:
I've attached the xacro-generated URDFs for the following models:
- wx250
- wx250s
- vx300
- vx300s

[urdfs.zip](https://github.com/Interbotix/interbotix_ros_manipulators/files/13551950/urdfs.zip)

Let me know if you need anything else.

kevinzakka, 2023-12-04:
Thank you for being so swift with the request, really appreciate it!

closed 2023-12-04 by kevinzakka

kevinzakka, 2023-12-05:
Hi @LSinterbotix, just want to confirm with you that the "s" in 300s corresponds to the 6dof model and the one without is the 5dof one?

LSinterbotix, 2023-12-05:
> the "s" in 300s corresponds to the 6dof model and the one without is the 5dof one

Correct — "s" for "six degrees of freedom".

kevinzakka, 2023-12-05:
Hi @LSinterbotix, the model is now live: https://github.com/google-deepmind/mujoco_menagerie/tree/main/trossen_vx300s

Thanks so much for your prompt help :)