Interbotix/interbotix_ros_manipulators #148: [Question]:

manishvel8, 2024-01-04:
### Question

We have a WX200 robotic arm. From that arm, we are playing chess. So for chess, we need coordinates for all 64 squares. So the arm has to move all 64 chess coordinates without exceeding the joint limit. In this, I will attach the four corners. chess board

a1 = (0.16, 0.17, 0.2)
a8 = (0.48, 0.17, 0.2)
h1 = (0.16, -0.17, 0.2)
h8 = (0.48, -0.17, 0.2)

From these four, we need coordinate values for all the 64-chess coordinates.

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2024-01-04:
Could you clarify exactly what you're asking here?

LSinterbotix, 2024-01-12:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2024-01-12 by LSinterbotix