Interbotix/interbotix_ros_manipulators #155: Reinforcement learning

kethan637, 2024-01-15:
### Question

Is that possible to configure reinforcement learning to wx200 robotic arm?
And also is any one done that before if you have those link can you share the link?
Your Use case for reinforcement learning for wx200 arm is to play chess game with that is that possible if possible please give me some references?

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

LSinterbotix, 2024-01-15:
I do not know of anyone that has done reinforcement learning for chess using our arms. Additionally, we do not have any reinforcement learning resources for our arms. The possibility of reinforcement learning has already been answered in #154 with the mention of [Google Deepmind's mujoco_menagerie](https://github.com/google-deepmind/mujoco_menagerie).

closed 2024-01-15 by LSinterbotix

kethan637, 2024-01-16:
but interbotix wx200 robo not there in that link right how to see your interbotix arm reinforcement learning

LSinterbotix, 2024-01-16:
> interbotix wx200 robo not there

Right, but the vx300s is. You may be able to set something up similar to that since the arms are so similar (minus a degree of freedom). We do not have much experience with [MuJoCo](https://mujoco.readthedocs.io/en/stable/overview.html) so you may want to look for some resources online. For example, [this page](https://mujoco.readthedocs.io/en/latest/modeling.html#curdf) describes modeling a scene and mentions URDFs. The [trossen_vx300s model xml](https://github.com/google-deepmind/mujoco_menagerie/blob/main/trossen_vx300s/vx300s.xml) can be a good starting point to make your own MuJoCo-based simulation.

kethan637, 2024-01-22:
Is that possible to move the interbotix wx200 arm with live coordinates from the camera vision opencv if possible can ou send some reference so that it is much helpful and also if live possible how to use roll and pitch  option

LSinterbotix, 2024-01-22:
Are you asking about [Visual Servoing](https://en.wikipedia.org/wiki/Visual_servoing)?

We don't have demos of that ourselves, but a quick Google search shows some promising results using [VISP](https://visp.inria.fr/):

- https://wiki.ros.org/visp_ros
- https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-bridge-ros.html

kethan637, 2024-01-24:
not about visual servoing I am asking then wx200 robotic arm will be moving x,y,z values in the pose_component  so my question is that possible to take the xyz values using live camera with opencv or not because in opencv we will not get the roll and pitch functionality for the chess game usecase so suggest me the reference say how to move the wx200 robotic arm with live camera coordiantes(or) is that possible to move robotic arm without xyz values if yes can you say how

LSinterbotix, 2024-01-24:
The arm comes with an AprilTag attached to it. You can use the [apriltag_ros package](https://github.com/AprilRobotics/apriltag_ros) or [our Python modules](https://github.com/Interbotix/interbotix_ros_toolboxes/tree/main/interbotix_perception_toolbox/interbotix_perception_modules/src/interbotix_perception_modules) built around it to get a very accurate pose estimate of the tag (and thus the arm) with respect to the camera.

kethan637, 2024-01-25:
Is Perception and live coordinates the same or not? because I am not getting clarity how to move to at particular of arm through the camera vision

kethan637, 2024-01-25:
And one more question is the possible to play chess game with wx200 arm with live camera coordinates or not if possible how to get through it send me some reference

kethan637, 2024-01-30:
can you please reply me for the above two comments

kethan637, 2024-02-01:
Hai, is there any working demo of interbotix wx200 arm other then chess and also can you say how to use inverse kinematics for the finding the coordinate values can you give me the reference for the inverse kinematics

LSinterbotix, 2024-02-01:
There are a few ways you can control the arm.

# Interbotix Python-ROS API

You can use our Python API to control the arm's joints, end effector, and gripper. See the [API documentation](https://docs.trossenrobotics.com/interbotix_xsarms_docs/python_ros_interface.html), [the demos](https://github.com/Interbotix/interbotix_ros_manipulators/tree/main/interbotix_ros_xsarms/examples/python_demos), and the [documentation for the demos](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros1_packages/python_demos.html). This has functions that call an iterative IK engine found in the modules in the [arm.py file here](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/main/interbotix_xs_toolbox/interbotix_xs_modules/src/interbotix_xs_modules/arm.py).

# MoveIt

We've set up a few demos on using MoveIt including:
- MoveIt's RViz Motion Planning Display launched by [the xsarm_moveit package](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros1_packages/moveit_motion_planning_configuration.html)
- MoveIt Interbotix Interface Demo GUI
- MoveGroup C++ API Interbotix Demo
- MoveGroup Python API Interbotix Demo

For info on the Interbotix demos, see [the xsarm_moveit_interface documentation](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros1_packages/moveit_interface_and_api.html).

kethan637, 2024-02-05:
We have interbotix wx 200 arm with 5dof. so is that possible to use inverse kinematics to our robotic arm.
 We found one code in interbotix but it is using 6DOF. Link - https://github.com/Interbotix/interbotix_ros_manipulators/blob/main/interbotix_ros_xsarms/examples/python_demos/ee_pose_matrix_control.py
 
 So, we want to know with 5 DOF inverse kinematics is possible or not. If possible give us some links to refer
 without inverse kinematic is that possible to get live coordinnates with camera
->We used perception also it is  like pick and place only where they are using place position with hardcode values can you explain how arm can reach a destination through the camera should pick and place in the destination using live camera

->can you send some references also..

LSinterbotix, 2024-02-05:
The robot_model arg for the InterbotixManipulatorXS can be changed for whatever model you're using. However, some of the demos have goals that can only be acheived using 6DoF arms like the one you're linking to. Others, like the [ee_pose_components demo](https://github.com/Interbotix/interbotix_ros_manipulators/blob/main/interbotix_ros_xsarms/examples/python_demos/ee_pose_components.py) do not have that limitation and can be used with your wx200. Simply change [the "wx250" in the object's constructor](https://github.com/Interbotix/interbotix_ros_manipulators/blob/main/interbotix_ros_xsarms/examples/python_demos/ee_pose_components.py#L9) to your robot model codename.

LSinterbotix, 2024-02-05:
> how arm can reach a destination through the camera should pick and place in the destination using live camera

We have some demo packages that use stereo depth camera output to find the location of objects within its field of view and command the arm to do simple pick-and-place tasks. See the [xsarm_perception package docs](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros1_packages/perception_pipeline_configuration.html) and the [demos](https://github.com/Interbotix/interbotix_ros_manipulators/tree/main/interbotix_ros_xsarms/interbotix_xsarm_perception/scripts) for details.

kethan637, 2024-02-06:
> We have some demo packages that use stereo depth camera output to find the location of objects within its field of view and command the arm to do simple pick-and-place tasks. See the [...]

->For that pick and place perception demo also have place value to hard coded value so can you say picking I am able to do with camera but placing they are giving the hardcode value so it is going there can yo say how to place in position through live camera