Interbotix/interbotix_ros_manipulators #158: [Question]: How to control the arm with the way of topic pub in gazebo？

EndlessLoops, 2024-01-25:
### Question

I use the wx200 in gazebo and I dont know how to control the arm with the way of topic pub. I find a [tutorial video](https://www.youtube.com/watch?v=k3zkgN7TYTE) in the youtube but the name and tpye of the arm topic on video is different from current code. 
[image]
current topic list：
```
$ rostopic list
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/performance_metrics
/gazebo/set_link_state
/gazebo/set_model_state
/rosout
/rosout_agg
/tf
/tf_static
/wx200/arm_controller/command
/wx200/arm_controller/follow_joint_trajectory/cancel
/wx200/arm_controller/follow_joint_trajectory/feedback
/wx200/arm_controller/follow_joint_trajectory/goal
/wx200/arm_controller/follow_joint_trajectory/result
/wx200/arm_controller/follow_joint_trajectory/status
/wx200/arm_controller/gains/elbow/parameter_descriptions
/wx200/arm_controller/gains/elbow/parameter_updates
/wx200/arm_controller/gains/shoulder/parameter_descriptions
/wx200/arm_controller/gains/shoulder/parameter_updates
/wx200/arm_controller/gains/waist/parameter_descriptions
/wx200/arm_controller/gains/waist/parameter_updates
/wx200/arm_controller/gains/wrist_angle/parameter_descriptions
/wx200/arm_controller/gains/wrist_angle/parameter_updates
/wx200/arm_controller/gains/wrist_rotate/parameter_descriptions
/wx200/arm_controller/gains/wrist_rotate/parameter_updates
/wx200/arm_controller/state
/wx200/gripper_controller/command
/wx200/gripper_controller/follow_joint_trajectory/cancel
/wx200/gripper_controller/follow_joint_trajectory/feedback
/wx200/gripper_controller/follow_joint_trajectory/goal
/wx200/gripper_controller/follow_joint_trajectory/result
/wx200/gripper_controller/follow_joint_trajectory/status
/wx200/gripper_controller/gains/left_finger/parameter_descriptions
/wx200/gripper_controller/gains/left_finger/parameter_updates
/wx200/gripper_controller/gains/right_finger/parameter_descriptions
/wx200/gripper_controller/gains/right_finger/parameter_updates
/wx200/gripper_controller/state
/wx200/joint_states
```

And I try to use this code but it seems to be a problem.What's wrong with this code?
`$ rostopic pub -1 /wx200/gripper_controller/command trajectory_msgs/JointTrajectory '{header: {seq: 1, stamp: {secs: 0, nsecs: 0}, frame_id: ""}, joint_names: ["left_finger", "right_finger"], points: [{positions: [1, 0], velocities: [1, 0.1], accelerations: [1, 0.1], effort: [], time_from_start: {secs: 1000, nsecs: 0}}]}'`
[image]

If I want to use the way of topic pub, how can I get it to stand up and open the clamp? Please give me aexample

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

wellyowo, 2024-01-30:
In the tutorial video, he used used_position_controller:=true to control the arm. And in your case you use trajectory contoller to control the arm. If you change to position_controller you will get the topic as the video. 
For right finger the value is -1 to 0 and for left finger the value is 0 to 1.

EndlessLoops, 2024-02-02:
thanks for your rely , I will try to test it after。

closed 2024-02-02 by EndlessLoops