Interbotix/interbotix_ros_manipulators #167: [Question]: Gazebo Simulation

manishvel8, 2024-02-07:
### Question

Hi Team,
How to operate the Interbotix WX200 arm in a Gazebo virtual simulation. Also, what files will be supported while importing a new model?

### Robot Model

wx200

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

manishvel8, 2024-02-08:
Also, tell us the steps for how to import the model into the Gazebo simulation.

LSinterbotix, 2024-02-13:
For details on launching the Gazebo demo package, see the [Gazebo configuration docs](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros1_packages/gazebo_simulation_configuration.html).

The arm launched by the Gazebo packages can be controlled using MoveIt. See the [MoveIt Configuration package docs](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros1_packages/moveit_motion_planning_configuration.html) and the [MoveIt Interface package docs](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros1_packages/moveit_interface_and_api.html) for details. Note that the Interbotix Python-ROS API is not compatible with robot simulated in Gazebo.

manishvel8, 2024-02-13:
So we launched a gazebo. In that gazebo, our interbotix arm is also moving. Now we want to import a tic-tac-toe board beside our arm in the gazebo simulation. How do I import tic-tac-toe boards in Gazebo? What files will be supported for importing in Gazebo?

manishvel8, 2024-02-16:
> So we launched a gazebo. In that gazebo, our interbotix arm is also moving. Now we want to import a tic-tac-toe board beside our arm in the gazebo simulation. How do I import tic-tac-toe boards in Gazebo? What files will be supported for importing in Gazebo?

@LSinterbotix  any link here?

LSinterbotix, 2024-02-16:
Please see the [Gazebo tutorials](https://classic.gazebosim.org/tutorials) to get answers to questions like these.

You'll first need to [create models](https://classic.gazebosim.org/tutorials?tut=build_model) for each unique component in your game and then either [create a world](https://classic.gazebosim.org/tutorials?cat=build_world) placing your object where needed, or spawn them in using the ROS-Gazebo [entity spawner](https://wiki.ros.org/simulator_gazebo/Tutorials/SpawningObjectInSimulation).

lukeschmitt-tr, 2024-02-21:
You should create a world that contains your model, or use the ROS-Gazebo entity spawner to spawn custom models into an existing world. See my previous comment for links.

manishvel8, 2024-02-23:
I'm asking which file will be supported for importing models in the gazebo virtual simulation. whether it is STL files or SDF files?

lukeschmitt-tr, 2024-02-23:
I believe Gazebo supports mesh geometry files in the formats STL, DAE, and OBJ. You can create an SDF file to describe your model, or you can spawn in a URDF-based robot description using their entity spawner. See their tutorials for more information.

lukeschmitt-tr, 2024-03-04:
Do you have any further questions on this? Otherwise, this issue will be closed.

closed 2024-03-08 by lukeschmitt-tr