Interbotix/interbotix_ros_manipulators #168: [Question]: How to uninstall amd packages

DaydreamPuppeh, 2024-02-07:
### Question

I'm attempting to install these packages on the jetson orin nano, and installed the amd version of the packages. I now realize the chip is aarch64, meaning I need to uninstall the amd version and install the arm version. How do I uninstall these packages. I rarely ever use linux so I don't know what to do. Also is the dynamixel wizard a requirement? The nano can't install it.

### Operating System

Ubuntu 20.04

### ROS Version

ROS 2 Galactic

LSinterbotix, 2024-02-13:
> How do I uninstall these packages

To remove the packages, you can delete the `~/interbotix_ws` directory and remove the configurations from your `~/.bashrc` file. However, the install scripts simply run a source build. If the installation went through without any issues, you should be able to use the packages and demos.

> is the dynamixel wizard a requirement? The nano can't install it.

This is not a requirement, it's just used for troubleshooting and can be run on other machines. The wizard tool is only built for x86 machines.

lukeschmitt-tr, 2024-02-21:
Do you have any further questions on this? Otherwise, this issue will be closed.

closed 2024-02-27 by lukeschmitt-tr