Interbotix/interbotix_ros_manipulators #169: [Question]: ModuleNotFoundError: No module named 'interbotix_xs_modules.arm'

kks32, 2024-02-14:
### Question

I'm able to install the interbotix packages using the instructions here https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/software_setup.html#interbotix-ros-packages. And can control the vx300s robot using the Rviz UI to go to home pose and sleep mode. However, when I run the bartender.py I get
```
Traceback (most recent call last):
  File "/home/krishna/Downloads/bartender.py", line 1, in <module>
    from interbotix_xs_modules.arm import InterbotixManipulatorXS
ModuleNotFoundError: No module named 'interbotix_xs_modules.arm'
```
Running
 > ros2 pkg list | grep interbotix

Gives me:
```
interbotix_common_modules
interbotix_common_sim
interbotix_common_toolbox
interbotix_moveit_interface
interbotix_moveit_interface_msgs
interbotix_perception_modules
interbotix_perception_msgs
interbotix_perception_pipelines
interbotix_perception_toolbox
interbotix_ros_xsarms
interbotix_ros_xsarms_examples
interbotix_ros_xseries
interbotix_tf_tools
interbotix_xs_driver
interbotix_xs_modules
interbotix_xs_msgs
interbotix_xs_ros_control
interbotix_xs_rviz
interbotix_xs_sdk
interbotix_xs_toolbox
interbotix_xsarm_control
interbotix_xsarm_descriptions
interbotix_xsarm_joy
interbotix_xsarm_moveit
interbotix_xsarm_moveit_interface
interbotix_xsarm_perception
interbotix_xsarm_ros_control
interbotix_xsarm_sim
```

I'm also running `ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=vx300s` in the other terminal window. 

I checked my python path::

```
 echo $PYTHONPATH
/home/krishna/interbotix_ws/install/interbotix_xs_modules/lib/python3.10/site-packages:/home/krishna/interbotix_ws/install/interbotix_perception_modules/lib/python3.10/site-packages:/home/krishna/interbotix_ws/install/interbotix_xs_msgs/local/lib/python3.10/dist-packages:/home/krishna/interbotix_ws/install/interbotix_perception_msgs/local/lib/python3.10/dist-packages:/home/krishna/interbotix_ws/install/interbotix_moveit_interface_msgs/local/lib/python3.10/dist-packages:/home/krishna/interbotix_ws/install/interbotix_common_modules/lib/python3.10/site-packages:/home/krishna/apriltag_ws/install/apriltag_ros/local/lib/python3.10/dist-packages:/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
```

Which seems to be pointing to the right modules.

### Robot Model

vx300s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

LSinterbotix, 2024-02-14:
Looks like you've downloaded the [bartender.py script from main](https://github.com/Interbotix/interbotix_ros_manipulators/blob/main/interbotix_ros_xsarms/examples/python_demos/bartender.py) or some other ROS 1 branch. The ROS 1 modules have a different module layout than our ROS 2 modules.

Note the different import paths:

ROS 1

https://github.com/Interbotix/interbotix_ros_manipulators/blob/341878954d92c7049d74eb8df2653f8012ea9dd1/interbotix_ros_xsarms/examples/python_demos/bartender.py#L1

ROS 2

https://github.com/Interbotix/interbotix_ros_manipulators/blob/777c2711f8c85523af096871ff3cc2c0dd78c61f/interbotix_ros_xsarms/interbotix_xsarm_control/demos/python_ros2_api/bartender.py#L33

Please use the demo scripts in [this directory](https://github.com/Interbotix/interbotix_ros_manipulators/tree/humble/interbotix_ros_xsarms/interbotix_xsarm_control/demos/python_ros2_api) if using ROS 2.

kks32, 2024-02-14:
Thank you. That was the problem. ROS2 bartender.py works fine!

closed 2024-02-14 by kks32