Interbotix/interbotix_ros_manipulators #174: [Question]: Is Dynamixel Trajectory Profiling Implemented for Simulated Controllers?

QuantuMope, 2024-03-04:
### Question

Hello. My team and I are working on training some sim2real policies on the WX250s arm.

Due to observing some very jerky policies being deployed, we decided to possibly start utilizing the trajectory profiling capabilities of the Dynamixel motors (i.e., setting profile velocity and profile acceleration $\neq$ 0).

But, upon observing the `interbotix_xsarm_sim` directory, I don't see any of Dynamixel's trajectory profiling logic being implemented on the sim side, which would undoubtedly lead to some sim2real gap.

Therefore, my questions are:
1. Is Dynamixel's trajectory profiling logic implemented for simulated controllers anywhere?
2. If not implemented, has it been assumed that sim2real researchers would stick to pure step response actuation?

Thanks in advance!

### Robot Model

wx250s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

lukeschmitt-tr, 2024-03-08:
The trajectory profiling logic is indeed not simulated in Gazebo. The sim package is really only there for simulating MoveIt or position trajectory-based control of the arm, not as much sim2real research. However, we're interested in spinning up other, better simulations if there is enough interest. For example, we've had some inquiries regarding Isaac Sim, MuJoCo, and OpenAI Gym. Our vx300s is [included in the MuJoCo Menagerie](https://github.com/google-deepmind/mujoco_menagerie/tree/main/trossen_vx300s) if you're interested in taking a look there.

QuantuMope, 2024-03-08:
Got it. Thanks!

closed 2024-03-08 by QuantuMope