Interbotix/interbotix_ros_manipulators #177: [Bug]: Four servos not working and detecting with ROS and Dynamixel Wizard

ncbdrck, 2024-03-28:
### What happened?

I have an RX200 robot arm, and one of the servos (Wrist Angle) was not working properly, flashing the red LED. I tried troubleshooting the issue by power cycling the motor according to the instructions, but unfortunately, the error continued. 

So, I tried connecting to it using the Dynamixel Wizard and did a scan to find the connected servos. During the scan, it found all the servos; however, they got disconnected automatically after a few minutes. 

Then, when I tried to connect again, it only detected the **three** `XM430-W350-T` servos but not the **four** `XL430-W250-T` servos.  

Also, now when I try connecting with ROS, I get the `Can't find DYNAMIXEL ID 'x', Joint Name : 'xxx' ` for the 4 missing servos.

### Robot Model

rx200

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 1 Noetic

### Steps To Reproduce

1. When the joint flashed red, I could temporarily get it working using the `reboot_motors` ROS service.
2. When connecting with Dynamixel Wizard for the first time to fix the faulty servo, I had to use both the `57600 bps` and `1000000 bps` to detect all the servos.
3. At the second round of checking, I checked all the options (protocols, ports, baud rates), but now it only detects three servos

### Relevant log output

```shell
NODES
  /rx200/
    controller_spawner (controller_manager/controller_manager)
    move_group (moveit_ros_move_group/move_group)
    moveit_python_interface (interbotix_moveit_interface/moveit_python_interface)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz_jay_Latitude_5310_6651_3709359738207999983 (rviz/rviz)
    xs_hardware_interface (interbotix_xs_ros_control/xs_hardware_interface)
    xs_sdk (interbotix_xs_sdk/xs_sdk)

auto-starting new master
process[master]: started with pid [6699]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 832d68e2-ed26-11ee-8bb2-8535b85a3a40
process[rosout-1]: started with pid [6720]
started core service [/rosout]
process[rx200/robot_state_publisher-2]: started with pid [6727]
process[rx200/xs_sdk-3]: started with pid [6728]
process[rx200/controller_spawner-4]: started with pid [6729]
process[rx200/xs_hardware_interface-5]: started with pid [6730]
process[rx200/move_group-6]: started with pid [6731]
process[rx200/rviz_jay_Latitude_5310_6651_3709359738207999983-7]: started with pid [6733]
[ INFO] [1711646017.414963376]: [xs_sdk] Loaded mode configs from '/home/jay/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_moveit_interface/config/modes.yaml'.
process[rx200/moveit_python_interface-8]: started with pid [6747]
[ INFO] [1711646017.417247048]: [xs_sdk] Loaded motor configs from '/home/jay/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_control/config/rx200.yaml'.
[ INFO] [1711646017.418100525]: [xs_sdk] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[ INFO] [1711646017.421894136]: waitForService: Service [/rx200/get_robot_info] has not been advertised, waiting...
[ERROR] [1711646017.486366428]: [xs_sdk]	Can't find DYNAMIXEL ID:  6, Joint Name: 'wrist_rotate':
		  '[TxRxResult] There is no status packet!'
[ WARN] [1711646017.513148593]: Falling back to using the move_group node's namespace (deprecated Melodic behavior).
[ INFO] [1711646017.527424788]: Loading robot model 'rx200'...
[ INFO] [1711646017.527513656]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ERROR] [1711646017.554859160]: [xs_sdk]	Can't find DYNAMIXEL ID:  5, Joint Name: 'wrist_angle':
		  '[TxRxResult] There is no status packet!'
[ INFO] [1711646017.591124282]: [xs_sdk]	Found DYNAMIXEL ID:  4, Model: 'XM430-W350', Joint Name: 'elbow'.
[ INFO] [1711646017.639110142]: [xs_sdk]	Found DYNAMIXEL ID:  2, Model: 'XM430-W350', Joint Name: 'shoulder'.
[ERROR] [1711646017.719281890]: [xs_sdk]	Can't find DYNAMIXEL ID:  7, Joint Name: 'gripper':
		  '[TxRxResult] There is no status packet!'
[... 82 lines cut ...]
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1711646051.969696978]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1711646051.970024234]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1711646051.970049094]: MoveGroup context initialization complete
```

lukeschmitt-tr, 2024-04-08:
@ncbdrck Please contact our support team at [this link](https://www.trossenrobotics.com/support) and cite this issue.

lukeschmitt-tr, 2024-04-19:
This issue was resolved through our support team. 3/4 motors needed firmware recovery. The last motor needed to be replaced.

closed 2024-04-19 by lukeschmitt-tr