Interbotix/interbotix_ros_manipulators #180: ROS2 Humble Installation

manishvel8, 2024-04-10:
### Question

Hi team,
How to install the ROS2 Humble version. We have an Interbotix WX200 robotic arm. Give a command to install. and give commands after installing how to launch the robotic arm in both a virtual and real-world arm. Also, in the ROS2 Gazebo simulation, will it work?

### Robot Model

wx200

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

lukeschmitt-tr, 2024-04-10:
> How to install the ROS2 Humble version

Please see this page for steps on installing ROS 2: https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/software_setup.html#software-installation

Here are the commands you would run to install the packages required for ROS 2 Humble:
```sh
sudo apt install curl
curl 'https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/main/interbotix_ros_xsarms/install/amd64/xsarm_amd64_install.sh' > xsarm_amd64_install.sh
chmod +x xsarm_amd64_install.sh
./xsarm_amd64_install.sh -d humble
```

> give commands after installing how to launch the robotic arm in both a virtual and real-world arm.

- real-world:
  - Launching: See the doc page for the arm control package: https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros2_packages/arm_control.html
  - Controlling:
    - See the Python demos doc page for Python-API demo usage: https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros2_packages/python_demos.html
    - See the MoveIt config pcakage doc page for MoveIt usage: https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros2_packages/moveit_motion_planning_configuration.html
- virtual: You have two options here:
  - Append `use_sim:=true` to the control launch command to launch a non-dynamic simulator for the DYNAMIXEL motors. You can use the Python demos mentioned above to control the arms this way.
  - See the doc page for the Gazebo simulator package: https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros2_packages/simulation_configuration.html. You can use MoveIt to control the arms this way. When using the MoveIt package, you can set the `hardware_type` arg to `gz_classic` to launch both Gazebo and MoveIt.

lukeschmitt-tr, 2024-04-16:
Do you have any further questions on this? Otherwise, this issue will be closed.