Interbotix/interbotix_ros_manipulators #182: Gazebo launch in ros2

kethan637, 2024-04-15:
### Question

i am not able to launch gazebo with robotic arm wx200 so when I tried to launch with these command ros2 launch interbotix_xsarm_sim xsarm_gz_classic.launch.py robot_model:=wx200 it is running and i able to see virtual arm in rviz but gazebo is not coming it is coming like this 
[INFO] [launch]: All log files can be found below /home/kethan/.ros/log/2024-04-15-16-56-26-033378-kethan-18049
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [18051]
[INFO] [gzclient-2]: process started with pid [18053]
[INFO] [spawn_entity.py-3]: process started with pid [18055]
[INFO] [robot_state_publisher-4]: process started with pid [18057]
[INFO] [rviz2-5]: process started with pid [18059]
[gzserver-1] [INFO] [1713180387.845118819] [gazebo_ros2_control]: Loading gazebo_ros2_control plugin
[gzserver-1] [INFO] [1713180387.847256596] [wx200.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in namespace: /wx200
[gzserver-1] [INFO] [1713180387.847389519] [wx200.gazebo_ros2_control]: Starting gazebo_ros2_control plugin in ros 2 node: gazebo_ros2_control
[gzserver-1] [INFO] [1713180387.849006374] [wx200.gazebo_ros2_control]: connected to service!! robot_state_publisher
[gzserver-1] [INFO] [1713180387.849319943] [wx200.gazebo_ros2_control]: Received urdf from param server, parsing...
[gzserver-1] [INFO] [1713180387.849341627] [wx200.gazebo_ros2_control]: Loading parameter files /home/kethan/interbotix_ws/install/interbotix_xsarm_sim/share/interbotix_xsarm_sim/config/trajectory_controllers/wx200_trajectory_controllers.yaml
[gzserver-1] [INFO] [1713180387.855606179] [wx200.gazebo_ros2_control]: Loading joint: waist
[gzserver-1] [INFO] [1713180387.855636783] [wx200.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1713180387.855643792] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855650912] [wx200.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1713180387.855655419] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855771192] [wx200.gazebo_ros2_control]: Loading joint: shoulder
[gzserver-1] [INFO] [1713180387.855775619] [wx200.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1713180387.855779982] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855784829] [wx200.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1713180387.855788651] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855798488] [wx200.gazebo_ros2_control]: Loading joint: elbow
[gzserver-1] [INFO] [1713180387.855802570] [wx200.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1713180387.855806161] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855809882] [wx200.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1713180387.855814409] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855823438] [wx200.gazebo_ros2_control]: Loading joint: wrist_angle
[gzserver-1] [INFO] [1713180387.855827371] [wx200.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1713180387.855831228] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855834787] [wx200.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1713180387.855838554] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855848672] [wx200.gazebo_ros2_control]: Loading joint: wrist_rotate
[gzserver-1] [INFO] [1713180387.855852918] [wx200.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1713180387.855857497] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855861516] [wx200.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1713180387.855959507] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855968395] [wx200.gazebo_ros2_control]: Loading joint: gripper
[gzserver-1] [INFO] [1713180387.855972577] [wx200.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1713180387.855976206] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855979827] [wx200.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1713180387.855983301] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.855988835] [wx200.gazebo_ros2_control]: Loading joint: left_finger
[gzserver-1] [INFO] [1713180387.855992547] [wx200.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1713180387.855997642] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.856002165] [wx200.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1713180387.856005723] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.856031202] [wx200.gazebo_ros2_control]: Loading joint: right_finger
[gzserver-1] [INFO] [1713180387.856035376] [wx200.gazebo_ros2_control]: 	State:
[gzserver-1] [INFO] [1713180387.856038952] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.856042542] [wx200.gazebo_ros2_control]: 	Command:
[gzserver-1] [INFO] [1713180387.856046062] [wx200.gazebo_ros2_control]: 	 position
[gzserver-1] [INFO] [1713180387.856080342] [resource_manager]: Initialize hardware 'XSHardwareInterface' 
[gzserver-1] [INFO] [1713180387.856165638] [resource_manager]: Successful initialization of hardware 'XSHardwareInterface'
[gzserver-1] [INFO] [1713180387.856232430] [resource_manager]: 'configure' hardware 'XSHardwareInterface' 
[gzserver-1] [INFO] [1713180387.856239216] [resource_manager]: Successful 'configure' of hardware 'XSHardwareInterface'
[gzserver-1] [INFO] [1713180387.856245010] [resource_manager]: 'activate' hardware 'XSHardwareInterface' 
[gzserver-1] [INFO] [1713180387.856279789] [resource_manager]: Successful 'activate' of hardware 'XSHardwareInterface'
[gzserver-1] [INFO] [1713180387.856336094] [wx200.gazebo_ros2_control]: Loading controller_manager
[gzserver-1] [INFO] [1713180387.866085024] [wx200.gazebo_ros2_control]: Loaded gazebo_ros2_control.
[INFO] [spawn_entity.py-3]: process has finished cleanly [pid 18055]
[INFO] [spawner-6]: process started with pid [18273]
[gzclient-2] 
[gzclient-2] libcurl: (6) Could not resolve host: fuel.ignitionrobotics.org
[gzserver-1] [INFO] [1713180388.162412406] [wx200.controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-6] [INFO] [1713180388.179883770] [wx200.joint_state_broadcaster_spawner]: Loaded joint_state_broadcaster
[... 20 lines cut ...]
[gzserver-1] [INFO] [1713180388.586259265] [wx200.arm_controller]: Command interfaces are [position] and state interfaces are [position].
[gzserver-1] [INFO] [1713180388.586264964] [wx200.arm_controller]: Using 'splines' interpolation method.
[gzserver-1] [INFO] [1713180388.586439739] [wx200.arm_controller]: Controller state will be published at 50.00 Hz.
[spawner-7] [INFO] [1713180388.586494164] [wx200.gripper_controller_spawner]: Configured and activated gripper_controller
[gzserver-1] [INFO] [1713180388.586839562] [wx200.arm_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-8] [INFO] [1713180388.593070304] [wx200.arm_controller_spawner]: Configured and activated arm_controller
[gzclient-2] gzclient: /usr/include/boost/smart_ptr/shared_ptr.hpp:728: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::rendering::Camera; typename boost::detail::sp_member_access<T>::type = gazebo::rendering::Camera*]: Assertion `px != 0' failed.
[INFO] [spawner-7]: process has finished cleanly [pid 18322]
[INFO] [spawner-8]: process has finished cleanly [pid 18324]
[ERROR] [gzclient-2]: process has died [pid 18053, exit code -6, cmd 'gzclient'].
[INFO] [rviz2-5]: process has finished cleanly [pid 18059]

can you say how to fix these and how to launch the gazebo in ros2 with wx200 arm

### Robot Model

wx200

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

kethan637, 2024-04-16:
[17:04] Siddharth Shorya
[spawner-6] [INFO] [1713267216.289611866] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist

[spawner-6] [INFO] [1713267218.297914504] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist

[INFO] [rviz2-5]: process has finished cleanly [pid 14059]

when I am trying to run gazebo gazebo opening but arm is not coming there

kethan637, 2024-04-16:
]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269543.213004734] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[INFO] [rviz2-5]: process has finished cleanly [pid 4655]
[spawner-6] [INFO] [1713269545.219252103] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269547.231365626] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269549.243726755] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269551.254708817] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269553.265316694] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269555.276335095] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269557.287315876] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269559.299260433] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269561.310862216] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[spawner-6] Traceback (most recent call last):
[spawner-6]   File "/opt/ros/humble/lib/controller_manager/spawner", line 33, in <module>
[spawner-6]     sys.exit(load_entry_point('controller-manager==2.40.0', 'console_scripts', 'spawner')())
[spawner-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 222, in main
[spawner-6]     if not wait_for_controller_manager(
[spawner-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 106, in wait_for_controller_manager
[spawner-6]     node_and_namespace = wait_for_value_or(
[spawner-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 66, in wait_for_value_or
[spawner-6]     time.sleep(0.2)
[spawner-6] KeyboardInterrupt
[ERROR] [spawner-6]: process has died [pid 4826, exit code -2, cmd '/opt/ros/humble/lib/controller_manager/spawner -c controller_manager joint_state_broadcaster --ros-args -r __node:=joint_state_broadcaster_spawner -r __ns:=/wx200 --params-file /tmp/launch_params_znuzhou3'].

not able to solve this error

lukeschmitt-tr, 2024-04-16:
Closed in favor of https://github.com/Interbotix/interbotix_ros_manipulators/issues/184