Interbotix/interbotix_ros_manipulators #184: gazebo

kethan637, 2024-04-16:
### Question

]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269543.213004734] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[INFO] [rviz2-5]: process has finished cleanly [pid 4655]
[spawner-6] [INFO] [1713269545.219252103] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269547.231365626] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269549.243726755] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269551.254708817] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269553.265316694] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269555.276335095] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269557.287315876] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269559.299260433] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
[spawner-6] [INFO] [1713269561.310862216] [wx200.joint_state_broadcaster_spawner]: Waiting for '/wx200/controller_manager' node to exist
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[spawner-6] Traceback (most recent call last):
[spawner-6]   File "/opt/ros/humble/lib/controller_manager/spawner", line 33, in <module>
[spawner-6]     sys.exit(load_entry_point('controller-manager==2.40.0', 'console_scripts', 'spawner')())
[spawner-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 222, in main
[spawner-6]     if not wait_for_controller_manager(
[spawner-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 106, in wait_for_controller_manager
[spawner-6]     node_and_namespace = wait_for_value_or(
[spawner-6]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/controller_manager/spawner.py", line 66, in wait_for_value_or
[spawner-6]     time.sleep(0.2)
[spawner-6] KeyboardInterrupt
[ERROR] [spawner-6]: process has died [pid 4826, exit code -2, cmd '/opt/ros/humble/lib/controller_manager/spawner -c controller_manager joint_state_broadcaster --ros-args -r __node:=joint_state_broadcaster_spawner -r __ns:=/wx200 --params-file /tmp/launch_params_znuzhou3'].

not able to fix this error arm is not launching in gazebo

### Robot Model

wx200

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

lukeschmitt-tr, 2024-04-16:
Please provide the launch argument you used for this and the entire log up to the errors.

kethan637, 2024-04-16:
ros2 launch interbotix_xsarm_sim xsarm_gz_classic.launch.py robot_model:=wx200
[INFO] [launch]: All log files can be found below /home/kethan/.ros/log/2024-04-16-20-26-23-426929-kethan-7411
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [7413]
[INFO] [gzclient-2]: process started with pid [7415]
[INFO] [spawn_entity.py-3]: process started with pid [7417]
[INFO] [robot_state_publisher-4]: process started with pid [7419]
[INFO] [rviz2-5]: process started with pid [7421]
[gzserver-1] Error Code 12 Msg: Unable to find uri[model://TrossenRoboticsOfficeBuilding]
[ERROR] [spawn_entity.py-3]: process has died [pid 7417, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity robot_description -topic wx200/robot_description -x 0.0 -y 0.0 -z 0.0 -Y 0.0 --ros-args -r __node:=spawn_wx200'].
[INFO] [spawner-6]: process started with pid [7666]
[spawner-6] [WARN] [1713279386.833630216] [wx200.joint_state_broadcaster_spawner]: Controller already loaded, skipping load_controller
[spawner-6] [ERROR] [1713279386.845438458] [wx200.joint_state_broadcaster_spawner]: Failed to configure controller
[ERROR] [spawner-6]: process has died [pid 7666, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner -c controller_manager joint_state_broadcaster --ros-args -r __node:=joint_state_broadcaster_spawner -r __ns:=/wx200 --params-file /tmp/launch_params_8muz0aep'].
[INFO] [spawner-7]: process started with pid [7692]
[INFO] [spawner-8]: process started with pid [7694]
[spawner-7] [WARN] [1713279388.016511190] [wx200.gripper_controller_spawner]: Controller already loaded, skipping load_controller
[spawner-8] [WARN] [1713279388.016601579] [wx200.arm_controller_spawner]: Controller already loaded, skipping load_controller
[spawner-8] [ERROR] [1713279388.032805866] [wx200.arm_controller_spawner]: Failed to configure controller
[spawner-7] [ERROR] [1713279388.032984540] [wx200.gripper_controller_spawner]: Failed to configure controller
[ERROR] [spawner-8]: process has died [pid 7694, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner -c controller_manager arm_controller --ros-args -r __node:=arm_controller_spawner -r __ns:=/wx200 --params-file /tmp/launch_params_cwwdijle'].
[ERROR] [spawner-7]: process has died [pid 7692, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner -c controller_manager gripper_controller --ros-args -r __node:=gripper_controller_spawner -r __ns:=/wx200 --params-file /tmp/launch_params_cen3snaj'].

lukeschmitt-tr, 2024-04-16:
Please run the sim launch command again, appending `verbose:=true`.

lukeschmitt-tr, 2024-04-23:
Do you have any further questions on this? Otherwise, this issue will be closed.

closed 2024-05-07 by lukeschmitt-tr