Interbotix/interbotix_ros_manipulators #193: Ros2 humble instaklation

kethan637, 2024-04-24:
### Question

CMake Error at CMakeLists.txt:21 (find_package):
  By not providing "Findinterbotix_xs_msgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "interbotix_xs_msgs", but CMake did not find one.

  Could not find a package configuration file provided by
  "interbotix_xs_msgs" with any of the following names:

    interbotix_xs_msgsConfig.cmake
    interbotix_xs_msgs-config.cmake

  Add the installation prefix of "interbotix_xs_msgs" to CMAKE_PREFIX_PATH or
  set "interbotix_xs_msgs_DIR" to a directory containing one of the above
  files.  If "interbotix_xs_msgs" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< interbotix_xs_ros_control [2.38s, exited with code 1]
Aborted  <<< interbotix_tf_tools [2.57s]                                                   
Aborted  <<< interbotix_moveit_interface_msgs [2.61s]                                      
Aborted  <<< interbotix_perception_msgs [2.72s]                                            
Aborted  <<< interbotix_xs_rviz [2.88s]                                             
Aborted  <<< interbotix_xsarm_descriptions [1.66s]                                  
Aborted  <<< moveit_visual_tools [5.65s]                             

Summary: 4 packages finished [6.00s]
  1 package failed: interbotix_xs_ros_control
  6 packages aborted: interbotix_moveit_interface_msgs interbotix_perception_msgs interbotix_tf_tools interbotix_xs_rviz interbotix_xsarm_descriptions moveit_visual_tools
  3 packages had stderr output: interbotix_xs_ros_control interbotix_xs_rviz moveit_visual_tools
  16 packages not processed
[ERROR] Failed to build Interbotix Arm ROS 2 Packages.
[ERROR] Interbotix Installation Failed!

getting this error while running the xsarm_amd64_install.sh

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

lukeschmitt-tr, 2024-04-25:
Try to delete the `~/interbotix_ws` directory and reattempt installation using the xsarm_amd64_install.sh script.

lukeschmitt-tr, 2024-05-07:
Do you have any further questions on this? Otherwise, this issue will be closed.

closed 2024-05-14 by lukeschmitt-tr