Interbotix/interbotix_ros_manipulators #195: [Question]:  Unable to contact ROS master at [http://<hostname>.local.11311

GRINGOLOCO7, 2024-05-20:
### Question

Hello, my name is Gregorio Orlando and I have the pleasure to work with one of your arm. In particular the wx250s.

I have a Raspberry PY, with hostname = ubuntu, connected to the arm. I followed the [RaspberryPi](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/raspberry_pi_setup.html) instructions to install Ubuntu and set upp the Pi. The I runned the 4 comands:

```
sudo apt install curl
curl 'https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/main/interbotix_ros_xsarms/install/rpi4/xsarm_rpi4_install.sh' > xsarm_rpi4_install.sh
chmod +x xsarm_rpi4_install.sh
./xsarm_rpi4_install.sh -d noetic
```

Everithing looks correct.

Then on my personla computer running ubuntu, I run these other comands for remote installation and controll arm from my computer, streaming trough the Pi.

```
sudo apt install curl
curl 'https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/main/interbotix_ros_xsarms/install/xsarm_remote_install.sh' > xsarm_remote_install.sh
chmod +x xsarm_remote_install.sh
./xsarm_remote_install.sh
```

Everithing looks correct.

Then I don't know how to run the usuals comands, and launch files, so that I muve the robot, but all computational operations (such as Rviz) append in  my personal computer.

MORE INFOS:
- In the Pi /dev/ I see the arm.
- Both computer are connected to same network trough ethernet cable. 
- In my personla computer installation, when I was asked to type hostname, I inserted "ubuntu". The name that comes out after typing in terminal of the RPi "hostname".
-  launching` roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=wx250s ` from from RPi, I start RViz in the pi. But also it is not able to connect to the robot. And says that cant locate node type [xs_sdk] in package [interbotics_xs_sdk]
- launching` roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=wx250s ` from from personal computer. I get error.  Unable to contact ROS master at [http://<hostname>.local.11311

### Robot Model

wx250s

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

lukeschmitt-tr, 2024-05-21:
> Unable to contact ROS master at [http://.local.11311

This makes it seem like a hostname was never set for the ROS_MASTER_URI environment variable. What is the output of running the command below:

```bash
printenv ROS_MASTER_URI
```

GRINGOLOCO7, 2024-05-22:
**I'll get to it tomorrow in the lab and check it out. Thank you for your comment.**

When I did the [remote installation](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/software_setup.html#remote-install) on my personal computer, I was asked to specify the hostname... Then in my case, the hostname of the Raspberry Pi is "ubuntu". 
I checked the`.bashrc` file, and it seems to update it correctly.
I also tried to swap "ubuntu" (hostname), directly with the IP address of my Raspberry Pi in the `.bashrc` file of my personal computer.

---------------------------------------------------------------------------
The error that pops out in the Raspberry Pi when I run:

`roslaunch interbotics_xsarm_control xsarm_control.launch robot_model:=wx250s use_rviz:=false`

is:

`ERROR: cannot launch node of type [interbotix_xs_sdk/xs_sdk]: Cannot locate node of type [xs_sdk] in package [interbotix_xs_sdk]. Make sure file exists in package path and permission is set to executable (chmod +x)`

And of coure when I run in the personal computer:

`roslaunch interbotics_xsarm_descriptions remove_view.launch `

I get similar error.

GRINGOLOCO7, 2024-05-24:
> > Unable to contact ROS master at [http://.local.11311
> 
> This makes it seem like a hostname was never set for the ROS_MASTER_URI environment variable. What is the output of running the command below:
> 
> ```shell
> printenv ROS_MASTER_URI
> ```

In the raspberry py, I get http://localhost:11311.
In the personal computer I get http://ubuntu.local:11311

GRINGOLOCO7, 2024-05-24:
in the pi:
```
roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=wx250s use_rviz:=false
... logging to /home/ubuntu/.ros/log/8fd30700-19e5-11ef-924b-dfaed8b34ae4/roslaunch-ubuntu-3892.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:36715/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /wx250s/robot_description: <?xml version="1....
 * /wx250s/xs_sdk/load_configs: True
 * /wx250s/xs_sdk/mode_configs: /home/ubuntu/inte...
 * /wx250s/xs_sdk/motor_configs: /home/ubuntu/inte...

NODES
  /wx250s/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    xs_sdk (interbotix_xs_sdk/xs_sdk)

ROS_MASTER_URI=http://localhost:11311

process[wx250s/robot_state_publisher-1]: started with pid [3909]
ERROR: cannot launch node of type [interbotix_xs_sdk/xs_sdk]: Cannot locate node of type [xs_sdk] in package [interbotix_xs_sdk]. Make sure file exists in package path and permission is set to executable (chmod +x)
[ WARN] [1716566410.633503741]: Shutdown request received.
[ WARN] [1716566410.637554432]: Reason given for shutdown: [[/wx250s/robot_state_publisher] Reason: new node registered with same name]
[wx250s/robot_state_publisher-1] process has finished cleanly
log file: /home/ubuntu/.ros/log/8fd30700-19e5-11ef-924b-dfaed8b34ae4/wx250s-robot_state_publisher-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

---------------------------------------------------------------

in the personal computer:

```
roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=wx250s
... logging to /home/gringo/.ros/log/8fd30700-19e5-11ef-924b-dfaed8b34ae4/roslaunch-helena-4681.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.205.1.25:38501/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /wx250s/robot_description: <?xml version="1....
 * /wx250s/xs_sdk/load_configs: True
 * /wx250s/xs_sdk/mode_configs: /home/gringo/inte...
 * /wx250s/xs_sdk/motor_configs: /home/gringo/inte...

NODES
  /wx250s/
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    xs_sdk (interbotix_xs_sdk/xs_sdk)

ROS_MASTER_URI=http://ubuntu.local:11311

process[wx250s/robot_state_publisher-1]: started with pid [4691]
process[wx250s/rviz-2]: started with pid [4692]
ERROR: cannot launch node of type [interbotix_xs_sdk/xs_sdk]: interbotix_xs_sdk
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/gringo/interbotix_ws/src
ROS path [2]=/opt/ros/noetic/share
[ WARN] [1716566410.645606866]: Shutdown request received.
[ WARN] [1716566410.648401969]: Reason given for shutdown: [[/wx250s/robot_state_publisher] Reason: new node registered with same name]
[wx250s/robot_state_publisher-1] process has finished cleanly
log file: /home/gringo/.ros/log/8fd30700-19e5-11ef-924b-dfaed8b34ae4/wx250s-robot_state_publisher-1*.log
[ERROR] [1716566411.166255083]: PluginlibFactory: The plugin for class 'interbotix_xs_rviz/Interbotix Control Panel' failed to load.  Error: According to the loaded plugin descriptions the class interbotix_xs_rviz/Interbotix Control Panel with base class type rviz::Panel does not exist. Declared types are  rviz_plugin_tutorials/Teleop
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command

```

lukeschmitt-tr, 2024-05-28:
# RPi

Was your workspace build? Try running `catkin_make` in the interbotix_ws root and check the output for errors. Make sure that you're also sourcing the workspace devel setup script.

# Remote

None of the packages are built when using the remote install script so you will not have access to any executables which is why the xsarm_control.launch command is failing.

However, running `roslaunch interbotics_xsarm_descriptions remove_view.launch` should not result in any errors. Could you share the error output there?

GRINGOLOCO7, 2024-06-13:
in RPi:
```
ubuntu@ubuntu:~$ echo $ROS_MASTER_URI 
http://localhost:11311
```

in Personal Computer:
```
gringo@cassandra:~$ echo $ROS_MASTER_URI 
http://10.205.1.131:11311
```
_(where 10.205.1.131 is IP of RPi)_
_(I also tryed with hostname as suggested in the .bashrc)_

--------------------------------------------------

in RPi:
`cd interbotix_ws `-> `catkin_make` => all good
`source interbotix_ws/devel/setup.bash`

in Personal Computer:
`cd interbotix_ws `-> `catkin_make` => 
```
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "interbotix_xs_sdk"
  with any of the following names:

    interbotix_xs_sdkConfig.cmake
    interbotix_xs_sdk-config.cmake

  Add the installation prefix of "interbotix_xs_sdk" to CMAKE_PREFIX_PATH or
  set "interbotix_xs_sdk_DIR" to a directory containing one of the above
  files.  If "interbotix_xs_sdk" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_control/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/gringo/interbotix_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/gringo/interbotix_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:320: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

`source interbotix_ws/devel/setup.bash`

---------------------------------------------------

in RPi:
`roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=wx250s use_rviz:=false`
=> [ INFO] [1718289299.815935126]: [xs_sdk] Interbotix 'xs_sdk' node is up!

in Personal Computer:
`roslaunch interbotix_xsarm_descriptions remote_view.launch `
```
RLException: Invalid <node> tag: <node> tag has an empty 'ns' attribute. 

Node xml is <node name="rviz" pkg="rviz" type="rviz" args="-d $(arg rvizconfig)" ns="$(arg robot_name)"/>
The traceback for the exception was written to the log file
```

lukeschmitt-tr, 2024-06-17:
# RPi

Looks like everything is working well on your RPi.

# Remote

The remote view launch requires that you provide a `robot_name` launch argument. This will be the same as the `robot_model` argument in the xsarm_control launch command as long as it is not overwritten. For the xsarm_control launch command you ran, the corresponding remote_view command would be:

```
roslaunch interbotics_xsarm_descriptions remove_view.launch robot_name:=wx250s
```

To your build errors — as stated previously, the remote installation script does not actually build any of the packages, including the interbotix_xs_sdk, which causes catkin_build to fail. This is intentional, as the remote only includes packages for remote viewing and simulation.

lukeschmitt-tr, 2024-07-08:
Do you have any further questions on this? Otherwise, this issue will be closed.

lukeschmitt-tr, 2024-07-16:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2024-07-16 by lukeschmitt-tr

GRINGOLOCO7, 2024-07-16:
Apology for late reply!!! I was busy these days. But yes, with your advices I were able to make everything working as expected. Thank you so much and kind regards. 
P.S. Congratulations, I really enjoy working with your arms and learning from you

lukeschmitt-tr, 2024-07-16:
No problem - happy to hear you got things up and running there. Appreciate your feedback as well!