Interbotix/interbotix_ros_manipulators #197: [Question]: The launch file is not able to find the world

ShayonMitra, 2024-05-21:
### Question

I am running this command: roslaunch xsarm_gazebo.launch in the directory ~/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_gazebo/launch . 

And I keep getting this error: Resource not found: The following package was not found in <arg name="world_name" default="$(find interbotix_xsarm_gazebo)/worlds/xsarm_gazebo.world"/>: interbotix_xsarm_gazebo
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/shayonmitra/interbotix_ws/src
ROS path [2]=/home/shayonmitra/catkin_ws/src
ROS path [3]=/opt/ros/noetic/share
The traceback for the exception was written to the log file

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

lukeschmitt-tr, 2024-07-08:
Do you have any further questions on this? Otherwise, this issue will be closed.

lukeschmitt-tr, 2024-07-16:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2024-07-16 by lukeschmitt-tr