Interbotix/interbotix_ros_manipulators #199: [Question]: ROS2 Humble install on RPi4

shonyred, 2024-05-28:
### Question

Hello,

I am having the same exact issue as in #193, although mine crashes when trying to build the interbotix_xs_modules package instead of the interbotix_xsarm_descriptions package. I am on a Raspberry Pi 4B as well, so I am using the xsarm_rpi4_install.sh script. I've already tried the suggestion there by deleting the script file and the interbotix_ws directory and re-downloading the script with curl (of course re-dowloading the rpi4 version), with no success. I've tried re-installing Ubuntu 22.04 multiple times as well. What's strange to me is that I've got it to download and install just fine on my desktop where I have Kubuntu 22.04, so maybe this is just an issue with the ARM64 script? 

Thank you.

[image]

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

lukeschmitt-tr, 2024-05-28:
Please see if the changes in #200 fix this issue.

Here's a link to the raw script file you can use in place of its standard location: https://raw.githubusercontent.com/Interbotix/interbotix_ros_manipulators/e901ef10845a86f3b6370ee60b0f3c036deffabb/interbotix_ros_xsarms/install/rpi4/xsarm_rpi4_install.sh

shonyred, 2024-05-28:
Thank you for the quick reply. I tried that new script and it installed fine. Thank you!

closed 2024-05-28 by shonyred