Interbotix/interbotix_ros_manipulators #204: The limits of axis in cartesian trajectory mode

barsm42, 2024-07-16:
### Question

Hello,

I am using simulation mode. The robot model is wx250s. I am sending the robot to different positions with bot.arm.set_ee_cartesian_trajectory function. I generate random numbers between different intervals. 

The position generation code is as follows:

x1 = random.randrange(-50,50)/100
y1 = random.randrange(-40,50)/100
z1 = random.randrange(-25,50)/100

Sometimes the robot goes to generated position, sometimes it doesn't go. The warning says the planned trajectory will not be executed.

I want to ask that is there any exact limit that I can move robot in x, y, and z axis without any errors? Between those limits, the robots should move always. I searched but could not find this kind of limits for 3 axis. 

### Robot Model

wx250s

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

closed 2024-07-16 by lukeschmitt-tr

barsm42, 2024-07-18:
Thank you. I arranged the limits.

lukeschmitt-tr, 2024-07-18:
We would be open to hosting this for others seeking the same information if you're willing to share your results.

barsm42, 2024-07-22:
`#!/usr/bin/env python3
# -*- coding: utf-8 -*-

from interbotix_xs_modules.arm import InterbotixManipulatorXS
import numpy as np
import sys, random
from time import sleep

def main():
    bot = InterbotixManipulatorXS("vx300s", "arm", "gripper")
    if (bot.arm.group_info.num_joints < 5):
        print('This demo requires the robot to have at least 5 joints!')
        sys.exit()
        
    while True:            
        bot.arm.go_to_home_pose()
        bot.arm.set_single_joint_position("wrist_angle", np.pi/2.0)
        pos = bot.arm.get_single_joint_command("wrist_angle")

        #"=======additional part"
        x1 = 0
        for k in range(0,100000,1):
            k = -k/100
            x1 = x1 +k
            bot.arm.set_ee_cartesian_trajectory(y=-0.01)
        #"=======end of the additional part"

if __name__=='__main__':
    main()

`
I added the "for" loop and decided the limits visually on simulation for x, y, and z axis separately for each direction. I will check the limits on real time, also.