Interbotix/interbotix_ros_manipulators #210: [Question]: How to use python interface to control gripper and joint at the same time

JiaheXu, 2024-08-10:
### Question

Hi developers,

I am using aloha system for research. In python demo scripts, the user has to specify the joints' position or the end-effector matrix for the manipulator, and the gripper is a separated part. To make the trajectory (joints and the gripper) move at the same time (in other to make the motion smoother), is there any way to do it?

### Robot Model

vp300s

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

lukeschmitt-tr, 2024-08-12:
By default, movement commands block execution of your program for some duration of time since the blocking parameter defaults to `True`. See an example of this [here](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/humble/interbotix_xs_toolbox/interbotix_xs_modules/interbotix_xs_modules/xs_robot/arm.py#L375). By setting this to `False` and manually blocking execution by sleeping or by some other method, both commands can be executed in parallel. You can see this done in ALOHA [here](https://github.com/Interbotix/aloha/blob/158b20628c8b6d3823dc86be6b5a4ce76d5604a4/aloha/real_env.py#L250-L251). You may also directly use the publishers that set joint and gripper commands as done [here](https://github.com/Interbotix/aloha/blob/158b20628c8b6d3823dc86be6b5a4ce76d5604a4/aloha/real_env.py#L175-L190).

lukeschmitt-tr, 2024-09-27:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2024-09-27 by lukeschmitt-tr