Interbotix/interbotix_ros_manipulators #211: [Question]: I built a new robot, how to do it without using gripper

parzivar, 2024-08-12:
### Question

:~/interbotix_ws$ ros2 launch interbotix_xsarm_ros_control xsarm_ros_control.launch.py robot_model:=master_neck2
[INFO] [launch]: All log files can be found below /home/bluelab/.ros/log/2024-08-12-10-40-37-332014-bluelab-320330
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [320332]
[INFO] [spawner-2]: process started with pid [320334]
[INFO] [xs_sdk-3]: process started with pid [320336]
[INFO] [robot_state_publisher-4]: process started with pid [320338]
[ros2_control_node-1] [WARN] [1723430437.701903155] [master_neck2.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-1] [INFO] [1723430437.702355355] [resource_manager]: Loading hardware 'XSHardwareInterface' 
[ros2_control_node-1] [INFO] [1723430437.706138287] [resource_manager]: Initialize hardware 'XSHardwareInterface' 
[xs_sdk-3] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.6'.
[xs_sdk-3] [INFO] Using logging level 'INFO'.
[xs_sdk-3] [INFO] Retrieving motor config file at '/home/bluelab/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/master_neck2.yaml'.
[xs_sdk-3] [INFO] Retrieving mode config file at '/home/bluelab/interbotix_ws/install/interbotix_xsarm_ros_control/share/interbotix_xsarm_ros_control/config/modes.yaml'.
[xs_sdk-3] [INFO] Loaded mode configs from '/home/bluelab/interbotix_ws/install/interbotix_xsarm_ros_control/share/interbotix_xsarm_ros_control/config/modes.yaml'.
[xs_sdk-3] [INFO] Loaded motor configs from '/home/bluelab/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/master_neck2.yaml'.
[xs_sdk-3] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[xs_sdk-3] [INFO] 	Found DYNAMIXEL ID:  4, Model: 'XM540-W270', Joint Name: 'wrist_angle'.
[xs_sdk-3] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XM540-W270', Joint Name: 'elbow'.
[xs_sdk-3] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XM540-W270', Joint Name: 'shoulder'.
[xs_sdk-3] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XM540-W270', Joint Name: 'waist'.
[xs_sdk-3] [WARN] Writing startup register values to EEPROM. This only needs to be done once on a robot if using a default motor config file, or after a motor config file has been modified. Can set `write_eeprom_on_startup` to false from now on.
[xs_sdk-3] [INFO] The operating mode for the 'arm' group was changed to 'position' with profile type 'velocity'.
[xs_sdk-3] [ERROR] The 'gripper' joint/group does not exist. Was it added to the motor config file?
[xs_sdk-3] [INFO] Interbotix X-Series Driver is up!
[xs_sdk-3] [ERROR] [1723430438.426531381] [interbotix_xs_sdk.xs_sdk]: Joint 'gripper' does not exist. Was it added to the motor config file?
[ros2_control_node-1] terminate called after throwing an instance of 'std::out_of_range'
[ros2_control_node-1]   what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[xs_sdk-3] [INFO] [1723430438.523870106] [interbotix_xs_sdk.xs_sdk]: InterbotixRobotXS is up!
[ERROR] [ros2_control_node-1]: process has died [pid 320332, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/master_neck2 --params-file /tmp/launch_params_uiq6lto0 --params-file /tmp/launch_params_e5ue_3g0'].
[spawner-2] [INFO] [1723430440.000149682] [master_neck2.arm_controller_spawner]: Waiting for '/master_neck2/controller_manager' services to be available
[spawner-2] [INFO] [1723430442.016349526] [master_neck2.arm_controller_spawner]: Waiting for '/master_neck2/controller_manager' services to be available
[spawner-2] [INFO] [1723430444.029253673] [master_neck2.arm_controller_spawner]: Waiting for '/master_neck2/controller_manager' services to be available
[spawner-2] [INFO] [1723430446.044529250] [master_neck2.arm_controller_spawner]: Waiting for '/master_neck2/controller_manager' services to be available
[spawner-2] [ERROR] [1723430448.060732071] [master_neck2.arm_controller_spawner]: Controller manager not available

### Robot Model

px100,I modified all the config based on px100

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

lukeschmitt-tr, 2024-08-12:
See if the steps described in https://github.com/Interbotix/interbotix_ros_manipulators/issues/75 answer this question for you.

parzivar, 2024-08-13:
thanks for your answer！！！now it can working such as run：
[image]
but when i run：ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=master_neck2 hardware_type:=fake
it can work, but will report errors：
[image]
And when I changed fake into actual, it couldn't work at all.
bluelab@bluelab:~/interbotix_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=master_neck2 hardware_type:=actual
[INFO] [launch]: All log files can be found below /home/bluelab/.ros/log/2024-08-13-11-05-28-979102-bluelab-467998
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [468002]
[INFO] [rviz2-2]: process started with pid [468004]
[INFO] [ros2_control_node-3]: process started with pid [468006]
[INFO] [spawner-4]: process started with pid [468008]
[INFO] [xs_sdk-5]: process started with pid [468010]
[INFO] [robot_state_publisher-6]: process started with pid [468012]
[xs_sdk-5] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.6'.
[xs_sdk-5] [INFO] Using logging level 'INFO'.
[xs_sdk-5] [INFO] Retrieving motor config file at '/home/bluelab/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/master_neck2.yaml'.
[xs_sdk-5] [INFO] Retrieving mode config file at '/home/bluelab/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/modes.yaml'.
[xs_sdk-5] [INFO] Loaded mode configs from '/home/bluelab/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/modes.yaml'.
[ros2_control_node-3] [WARN] [1723518329.635445783] [master_neck2.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1723518329.635645835] [resource_manager]: Loading hardware 'XSHardwareInterface' 
[xs_sdk-5] [INFO] Loaded motor configs from '/home/bluelab/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/master_neck2.yaml'.
[xs_sdk-5] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[ros2_control_node-3] [INFO] [1723518329.637676769] [resource_manager]: Initialize hardware 'XSHardwareInterface' 
[move_group-1] [WARN] [1723518329.652026722] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1723518329.654006628] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00181995 seconds
[move_group-1] [INFO] [1723518329.654037240] [moveit_robot_model.robot_model]: Loading robot model 'master_neck2'...
[move_group-1] [INFO] [1723518329.654043735] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[xs_sdk-5] [INFO] 	Found DYNAMIXEL ID:  4, Model: 'XM540-W270', Joint Name: 'wrist_angle'.
[xs_sdk-5] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XM540-W270', Joint Name: 'elbow'.
[xs_sdk-5] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XM540-W270', Joint Name: 'shoulder'.
[move_group-1] [INFO] [1723518329.777338446] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1723518329.777478449] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/master_neck2/joint_states' for joint states
[move_group-1] [INFO] [1723518329.778312951] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/master_neck2/joint_states'
[move_group-1] [INFO] [1723518329.778944038] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1723518329.778962007] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1723518329.779398822] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1723518329.779417130] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1723518329.779882022] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1723518329.780360730] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1723518329.780694409] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1723518329.780713089] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1723518329.780724043] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1723518329.804702099] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[xs_sdk-5] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XM540-W270', Joint Name: 'waist'.
[move_group-1] [INFO] [1723518329.815348670] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1723518329.817089887] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[... cut ...]
[spawner-4] [ERROR] [1723518339.963909699] [master_neck2.arm_controller_spawner]: Controller manager not available
[ERROR] [spawner-4]: process has died [pid 468008, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner -c /master_neck2/controller_manager arm_controller --ros-args -r __node:=arm_controller_spawner -r __ns:=/master_neck2'].

I have changed the URDF 、interbotix_ros_control/config、interbotix_xsarm_moveit/config， but cant solve the error： “Waiting for '/master_neck2/controller_manager' services to be available”

lukeschmitt-tr, 2024-08-14:
As of now, the [xs_hardware_interface](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/humble/interbotix_xs_toolbox/interbotix_xs_ros_control/src/xs_hardware_interface.cpp) (our hardware interface for ros2_control) expects that a gripper is part of the arm. You will have to modify that interface to remove that assumption. I would guess that the line causing the interface to crash is [this one](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/daae95ca3a5952e3e1d993e4900ab40c3ede8ec8/interbotix_xs_toolbox/interbotix_xs_ros_control/src/xs_hardware_interface.cpp#L90).

parzivar, 2024-08-15:
Thank you very much for your help. Now I can run actual customized robot!
Finally, I still have a problem. I use moveit for motion planning. When I check collision-aware IK, the robot cannot directly drag its end.
[image]

Below is my xs_hardware_interface.ccp file. I hope it can help others in need.

```
// Copyright 2022 Trossen Robotics
//
// Redistribution and use in source and binary forms, with or without
// modification, are permitted provided that the following conditions are met:
//
//    * Redistributions of source code must retain the above copyright
//      notice, this list of conditions and the following disclaimer.
//
//    * Redistributions in binary form must reproduce the above copyright
//      notice, this list of conditions and the following disclaimer in the
//      documentation and/or other materials provided with the distribution.
//
//    * Neither the name of the the copyright holder nor the names of its
//      contributors may be used to endorse or promote products derived from
//      this software without specific prior written permission.
//
// THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
// AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
// IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
// ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
// LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
// CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
// SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
// INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
// CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
// ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
// POSSIBILITY OF SUCH DAMAGE.

#include <interbotix_xs_ros_control/xs_hardware_interface.hpp>

#include <limits>
#include <memory>
#include <string>
#include <vector>

namespace interbotix_xs_ros_control
{

CallbackReturn XSHardwareInterface::on_init(const hardware_interface::HardwareInfo & info)
{
  info_ = info;

  nh = std::make_shared<rclcpp::Node>("xs_hardware_interface");
  executor = std::make_shared<rclcpp::executors::MultiThreadedExecutor>();
  executor->add_node(nh);

  // get hardware parameters
  group_name = info_.hardware_parameters["group_name"];
//  gripper_name = info_.hardware_parameters["gripper_name"];
  std::string js_topic = info_.hardware_parameters["joint_states_topic"];

  // create pubs, subs, and services
  pub_group = nh->create_publisher<interbotix_xs_msgs::msg::JointGroupCommand>(
    "commands/joint_group",
    1);
//  pub_gripper = nh->create_publisher<interbotix_xs_msgs::msg::JointSingleCommand>(
//    "commands/joint_single",
//    1);
  sub_joint_states = nh->create_subscription<sensor_msgs::msg::JointState>(
    js_topic,
    1,
    std::bind(&XSHardwareInterface::joint_state_cb, this, std::placeholders::_1));
  srv_robot_info = nh->create_client<interbotix_xs_msgs::srv::RobotInfo>("get_robot_info");
[... 151 lines cut ...]
  // Only publish commands if different than the previous update's commands
  if (joint_commands_prev != group_msg.cmd) {
    pub_group->publish(group_msg);
    joint_commands_prev = group_msg.cmd;
  }
//  if (gripper_cmd_prev != gripper_msg.cmd) {
//    pub_gripper->publish(gripper_msg);
//    gripper_cmd_prev = gripper_msg.cmd;
//  }
  return return_type::OK;
}

void XSHardwareInterface::joint_state_cb(const sensor_msgs::msg::JointState & msg)
{
  std::lock_guard<std::mutex> lck(joint_state_mtx_);
  joint_states = msg;
}

}  // namespace interbotix_xs_ros_control

#include "pluginlib/class_list_macros.hpp"  // NOLINT
PLUGINLIB_EXPORT_CLASS(
  interbotix_xs_ros_control::XSHardwareInterface,
  hardware_interface::SystemInterface)
```