Interbotix/interbotix_ros_manipulators #212: [Question]: Waiting for '/master_neck2/controller_manager' services to be available

parzivar, 2024-08-14:
### Question

First, I defined a 4-degree-of-freedom robot，I changed interbotix_xsarm_control/config/robot.yaml + mode.yaml files
then, I changed interbotix_xsarm_descriptions/meshes + urdf
so I can run: ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=master_neck2
and The real robo and the simulation look very well connected.
then，I changed the files in interbotix_xsarm_moveit， 
and run:ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=master_neck2 hardware_type:=fake    
This command works fine, but changing "fake" to "actual" doesn't work
Finaly,  I changed interbotix_xsarm_ros_control/config/controllers + modes.yaml . my controllers.yaml can be found in adtional info
The error report can be seen in additional info too.
I'm very much looking forward to your help analyzing this issue. thank you very much!

### Robot Model

px100, a customize robot based on px100

### Operating System

Ubuntu 22.04

### ROS Version

ROS 2 Humble

### Additional Info

bluelab@bluelab:~/interbotix_ws$ ros2 launch interbotix_xsarm_ros_control xsarm_ros_control.launch.py robot_model:=master_neck2
[INFO] [launch]: All log files can be found below /home/bluelab/.ros/log/2024-08-14-17-56-15-338930-bluelab-582539
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [582541]
[INFO] [spawner-2]: process started with pid [582543]
[INFO] [xs_sdk-3]: process started with pid [582545]
[INFO] [robot_state_publisher-4]: process started with pid [582547]
[ros2_control_node-1] [WARN] [1723629375.696091216] [master_neck2.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-1] [INFO] [1723629375.696239331] [resource_manager]: Loading hardware 'XSHardwareInterface' 
[ros2_control_node-1] [INFO] [1723629375.697937873] [resource_manager]: Initialize hardware 'XSHardwareInterface' 
[xs_sdk-3] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.6'.
[xs_sdk-3] [INFO] Using logging level 'INFO'.
[xs_sdk-3] [INFO] Retrieving motor config file at '/home/bluelab/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/master_neck2.yaml'.
[xs_sdk-3] [INFO] Retrieving mode config file at '/home/bluelab/interbotix_ws/install/interbotix_xsarm_ros_control/share/interbotix_xsarm_ros_control/config/modes.yaml'.
[xs_sdk-3] [INFO] Loaded mode configs from '/home/bluelab/interbotix_ws/install/interbotix_xsarm_ros_control/share/interbotix_xsarm_ros_control/config/modes.yaml'.
[xs_sdk-3] [INFO] Loaded motor configs from '/home/bluelab/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/master_neck2.yaml'.
[xs_sdk-3] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[xs_sdk-3] [INFO] 	Found DYNAMIXEL ID:  4, Model: 'XM540-W270', Joint Name: 'wrist_angle'.
[xs_sdk-3] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XM540-W270', Joint Name: 'elbow'.
[xs_sdk-3] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XM540-W270', Joint Name: 'shoulder'.
[xs_sdk-3] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XM540-W270', Joint Name: 'waist'.
[xs_sdk-3] [INFO] Skipping Load Configs.
[xs_sdk-3] [INFO] The operating mode for the 'arm' group was changed to 'position' with profile type 'velocity'.
[xs_sdk-3] [INFO] Interbotix X-Series Driver is up!
[xs_sdk-3] [ERROR] [1723629376.157926449] [interbotix_xs_sdk.xs_sdk]: Joint 'gripper' does not exist. Was it added to the motor config file?
[ros2_control_node-1] terminate called after throwing an instance of 'std::out_of_range'
[ros2_control_node-1]   what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[xs_sdk-3] [INFO] [1723629376.255543758] [interbotix_xs_sdk.xs_sdk]: InterbotixRobotXS is up!
[ERROR] [ros2_control_node-1]: process has died [pid 582541, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/master_neck2 --params-file /tmp/launch_params_d8wj_mbd --params-file /tmp/launch_params__4roaf3q'].
[spawner-2] [INFO] [1723629378.019945257] [master_neck2.arm_controller_spawner]: Waiting for '/master_neck2/controller_manager' services to be available
[spawner-2] [INFO] [1723629380.035532910] [master_neck2.arm_controller_spawner]: Waiting for '/master_neck2/controller_manager' services to be available
[spawner-2] [INFO] [1723629382.050814444] [master_neck2.arm_controller_spawner]: Waiting for '/master_neck2/controller_manager' services to be available
[spawner-2] [INFO] [1723629384.067606380] [master_neck2.arm_controller_spawner]: Waiting for '/master_neck2/controller_manager' services to be available
[spawner-2] [ERROR] [1723629386.082544616] [master_neck2.arm_controller_spawner]: Controller manager not available
[ERROR] [spawner-2]: process has died [pid 582543, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner -c /master_neck2/controller_manager arm_controller --ros-args -r __node:=arm_controller_spawner -r __ns:=/master_neck2'].

my controller.yaml :
/$(var robot_name):
  controller_manager:
    ros__parameters:
      update_rate: 10 # Hz
      arm_controller:
        type: joint_trajectory_controller/JointTrajectoryController
#      gripper_controller:
#        type: joint_trajectory_controller/JointTrajectoryController
      joint_state_broadcaster:
        type: joint_state_broadcaster/JointStateBroadcaster
  arm_controller:
    ros__parameters:
      joints:
        - waist
        - shoulder
        - elbow
        - wrist_angle

      command_interfaces:
        - position
      state_interfaces:
        - position
      state_publish_rate: 50.0 # Hz
      action_monitor_rate: 20.0 # Hz
      constraints:
        stopped_velocity_tolerance: 0.02
        goal_time: 1.0
#  gripper_controller:
#    ros__parameters:
#      joints:
#        - left_finger
#      command_interfaces:
#        - position
#      state_interfaces:
#        - position
  joint_state_broadcaster:
    ros__parameters:
      joints:
        - waist
        - shoulder
        - elbow
        - wrist_angle

parzivar, 2024-08-14:
Sorry, there are some characters in strange formats that I annotated in the original file.

lukeschmitt-tr, 2024-08-14:
Please don't open duplicate issues. See my response in [#211](https://github.com/Interbotix/interbotix_ros_manipulators/issues/211#issuecomment-2289334741).

closed 2024-08-14 by lukeschmitt-tr