Interbotix/interbotix_ros_manipulators #219: [Bug]: gazebo fails to launch

YumTaha, 2024-09-12:
### What happened?

I was following the MoveIt 2 planning configuration instructions in the ROS2 documentation. The command successfully launches MoveIt, but when it tries to open Gazebo, it fails and an error message saying ```process has died``` appears in the terminal. The terminal then repeatedly shows waiting for ```'/vx300/controller_manager' node to exist```

command used: ```ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=vx300 hardware_type:=gz_classic```

### Robot Model

vx300

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 2 Galactic

### Relevant log output

```shell
pibot@pibot:~$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=vx300 hardware_type:=gz_classic
[INFO] [launch]: All log files can be found below /home/pibot/.ros/log/2024-09-12-11-33-30-115523-pibot-3935
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [3945]
[INFO] [rviz2-2]: process started with pid [3947]
[INFO] [gzserver-3]: process started with pid [3949]
[INFO] [gzclient   -4]: process started with pid [3951]
[INFO] [spawn_entity.py-5]: process started with pid [3954]
[INFO] [robot_state_publisher-6]: process started with pid [3957]
[move_group-1] [WARN] [1726140818.294267076] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1726140818.294638182] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1726140818.351760697] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1726140818.352067341] [moveit_robot_model.robot_model]: Loading robot model 'vx300'...
[move_group-1] [INFO] [1726140818.352150377] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] Link vx300/base_link had 1 children
[move_group-1] Link vx300/shoulder_link had 1 children
[move_group-1] Link vx300/upper_arm_link had 1 children
[move_group-1] Link vx300/forearm_link had 1 children
[move_group-1] Link vx300/wrist_link had 1 children
[move_group-1] Link vx300/gripper_link had 1 children
[move_group-1] Link vx300/ee_arm_link had 2 children
[move_group-1] Link vx300/gripper_prop_link had 0 children
[move_group-1] Link vx300/gripper_bar_link had 1 children
[move_group-1] Link vx300/fingers_link had 3 children
[move_group-1] Link vx300/ee_gripper_link had 0 children
[move_group-1] Link vx300/left_finger_link had 0 children
[move_group-1] Link vx300/right_finger_link had 0 children
[move_group-1] [INFO] [1726140819.117157309] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1726140819.118091740] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/vx300/joint_states' for joint states
[move_group-1] [INFO] [1726140819.122145685] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/vx300/joint_states'
[move_group-1] [INFO] [1726140819.127444890] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1726140819.127586611] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1726140819.134209316] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1726140819.134327630] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1726140819.138904530] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1726140819.144306826] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1726140819.148415511] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1726140819.148654637] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1726140819.148773617] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1726140819.365306583] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1726140819.594210936] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1726140819.620100026] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1726140819.620204747] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1726140819.620233543] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1726140819.620305246] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1726140819.620372560] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[move_group-1] [INFO] [1726140819.620400430] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1726140819.620452744] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1726140819.620478743] [moveit_ros.fix_start_state_collision]: Param 'move_group.jiggle_fraction' was not set. Using default value: 0.020000
[... 68 lines cut ...]
[move_group-1] [INFO] [1726140852.822463177] [rclcpp]: signal_handler(signal_value=2)
[spawner-7] Traceback (most recent call last):
[spawner-7]   File "/opt/ros/galactic/lib/controller_manager/spawner", line 20, in <module>
[spawner-7]     ret = main()
[spawner-7]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 162, in main
[spawner-7]     if not wait_for_controller_manager(node, controller_manager_name,
[spawner-7]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 95, in wait_for_controller_manager
[spawner-7]     node_and_namespace = wait_for_value_or(
[spawner-7]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 57, in wait_for_value_or
[spawner-7]     time.sleep(0.2)
[spawner-7] KeyboardInterrupt
[move_group-1] [WARN] [1726140853.087403689] [moveit_ros.current_state_monitor]: No state update received within 100ms of system clock
[INFO] [robot_state_publisher-6]: process has finished cleanly [pid 3957]
[ERROR] [spawner-7]: process has died [pid 4070, exit code -2, cmd '/opt/ros/galactic/lib/controller_manager/spawner -c vx300/controller_manager joint_state_broadcaster --ros-args -r __node:=joint_state_broadcaster_spawner -r __ns:=/vx300 --params-file /tmp/launch_params_5vldg8ol'].
[move_group-1] [WARN] [1726140854.088592480] [moveit_ros.current_state_monitor]: No state update received within 100ms of system clock
[ERROR] [gzserver-3]: process has died [pid 3949, exit code -2, cmd 'gzserver /home/pibot/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world                                                                      -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[ERROR] [rviz2-2]: process has died [pid 3947, exit code -6, cmd '/opt/ros/galactic/lib/rviz2/rviz2 -d /home/pibot/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/rviz/xsarm_moveit.rviz -f world --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_uuwfivo1 --params-file /tmp/launch_params_7wajs1cn --params-file /tmp/launch_params_vy_ogkap --params-file /home/pibot/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/kinematics.yaml --params-file /tmp/launch_params_volfo3qw -r vx300/get_planning_scene:=/vx300/get_planning_scene -r /arm_contro [... cut ...] mp/launch_params_5zczd4ck --params-file /tmp/launch_params_ef_ptkym --params-file /tmp/launch_params_yo7555z6 --params-file /tmp/launch_params_zq55e80n --params-file /tmp/launch_params_b2serdvl --params-file /tmp/launch_params_63ic22km -r vx300/get_planning_scene:=/vx300/get_planning_scene -r /arm_controller/follow_joint_trajectory:=/vx300/arm_controller/follow_joint_trajectory -r /gripper_controller/follow_joint_trajectory:=/vx300/gripper_controller/follow_joint_trajectory'].
[move_group-1]
```

lukeschmitt-tr, 2024-09-12:
Please see if the answer below solves your issue:

https://github.com/Interbotix/interbotix_ros_manipulators/issues/182#issuecomment-2057188874

YumTaha, 2024-09-13:
Sourcing Gazebo fixes its issue not launching but I still cannot connect to the actual arm
using ```verbose:=true``` doesn't fix it either.

```
pibot@pibot:~/interbotix_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=vx300 hardware_type:=gz_classic
[INFO] [launch]: All log files can be found below /home/pibot/.ros/log/2024-09-13-12-44-35-972006-pibot-3227
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [3235]
[INFO] [rviz2-2]: process started with pid [3237]
[INFO] [gzserver-3]: process started with pid [3239]
[INFO] [gzclient   -4]: process started with pid [3241]
[INFO] [spawn_entity.py-5]: process started with pid [3245]
[INFO] [robot_state_publisher-6]: process started with pid [3247]
[move_group-1] [WARN] [1726231484.307555604] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1726231484.307827485] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1726231484.344718271] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1726231484.345132212] [moveit_robot_model.robot_model]: Loading robot model 'vx300'...
[move_group-1] [INFO] [1726231484.345180318] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] Link vx300/base_link had 1 children
[move_group-1] Link vx300/shoulder_link had 1 children
[move_group-1] Link vx300/upper_arm_link had 1 children
[move_group-1] Link vx300/forearm_link had 1 children
[move_group-1] Link vx300/wrist_link had 1 children
[move_group-1] Link vx300/gripper_link had 1 children
[move_group-1] Link vx300/ee_arm_link had 2 children
[move_group-1] Link vx300/gripper_prop_link had 0 children
[move_group-1] Link vx300/gripper_bar_link had 1 children
[move_group-1] Link vx300/fingers_link had 3 children
[move_group-1] Link vx300/ee_gripper_link had 0 children
[move_group-1] Link vx300/left_finger_link had 0 children
[move_group-1] Link vx300/right_finger_link had 0 children
[move_group-1] [INFO] [1726231485.218745857] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1726231485.223794347] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/vx300/joint_states' for joint states
[move_group-1] [INFO] [1726231485.232601377] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/vx300/joint_states'
[move_group-1] [INFO] [1726231485.239815155] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1726231485.242938721] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1726231485.254498797] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1726231485.257459860] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1726231485.263325140] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1726231485.271655754] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1726231485.291885281] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1726231485.292247061] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1726231485.292406379] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1726231485.444712584] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1726231485.658585044] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1726231485.685617501] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1726231485.685798650] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1726231485.685830851] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1726231485.685929841] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1726231485.686004871] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[move_group-1] [INFO] [1726231485.686034535] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1726231485.686089122] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[... cut ...]
[spawner-7] [INFO] [1726231498.331029360] [vx300.joint_state_broadcaster_spawner]: Waiting for '/vx300/controller_manager' node to exist
[spawner-7] [INFO] [1726231500.372638602] [vx300.joint_state_broadcaster_spawner]: Waiting for '/vx300/controller_manager' node to exist
[spawner-7] [INFO] [1726231502.460090249] [vx300.joint_state_broadcaster_spawner]: Waiting for '/vx300/controller_manager' node to exist
[spawner-7] [INFO] [1726231504.521014468] [vx300.joint_state_broadcaster_spawner]: Waiting for '/vx300/controller_manager' node to exist
```

YumTaha, 2024-09-13:
Nothing changed

```
pibot@pibot:~/interbotix_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=vx300 hardware_type:=gz_classic
[INFO] [launch]: All log files can be found below /home/pibot/.ros/log/2024-09-13-11-35-28-060806-pibot-4422
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [4430]
[INFO] [rviz2-2]: process started with pid [4432]
[INFO] [gzserver-3]: process started with pid [4434]
[INFO] [gzclient   -4]: process started with pid [4436]
[INFO] [spawn_entity.py-5]: process started with pid [4439]
[INFO] [robot_state_publisher-6]: process started with pid [4442]
[move_group-1] [WARN] [1726241736.862265168] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1726241736.862499851] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1726241736.882357158] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1726241736.882555711] [moveit_robot_model.robot_model]: Loading robot model 'vx300'...
[move_group-1] [INFO] [1726241736.882625507] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] Link vx300/base_link had 1 children
[move_group-1] Link vx300/shoulder_link had 1 children
[move_group-1] Link vx300/upper_arm_link had 1 children
[move_group-1] Link vx300/forearm_link had 1 children
[move_group-1] Link vx300/wrist_link had 1 children
[move_group-1] Link vx300/gripper_link had 1 children
[move_group-1] Link vx300/ee_arm_link had 2 children
[move_group-1] Link vx300/gripper_prop_link had 0 children
[move_group-1] Link vx300/gripper_bar_link had 1 children
[move_group-1] Link vx300/fingers_link had 3 children
[move_group-1] Link vx300/ee_gripper_link had 0 children
[move_group-1] Link vx300/left_finger_link had 0 children
[move_group-1] Link vx300/right_finger_link had 0 children
[move_group-1] [INFO] [1726241737.403830754] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1726241737.404867188] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/vx300/joint_states' for joint states
[move_group-1] [INFO] [1726241737.407975562] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/vx300/joint_states'
[move_group-1] [INFO] [1726241737.411541246] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1726241737.411650819] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1726241737.416837375] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1726241737.416940485] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1726241737.421547732] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1726241737.425207397] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1726241737.429231742] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1726241737.429389667] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1726241737.429491703] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1726241737.566855717] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1726241737.732554040] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1726241737.769966473] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1726241737.770085546] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1726241737.770115472] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1726241737.770205452] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1726241737.770276285] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[move_group-1] [INFO] [1726241737.770303748] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1726241737.770358302] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[... cut ...]
[ERROR] [spawn_entity.py-5]: process has died [pid 4439, exit code 1, cmd '/opt/ros/galactic/lib/gazebo_ros/spawn_entity.py -entity robot_description -topic vx300/robot_description -x 0.0 -y 0.0 -z 0.0 -Y 0.0 --ros-args -r __node:=spawn_vx300'].
[INFO] [spawner-7]: process started with pid [4559]
[spawner-7] [INFO] [1726241748.434822681] [vx300.joint_state_broadcaster_spawner]: Waiting for '/vx300/controller_manager' node to exist
[spawner-7] [INFO] [1726241750.490983763] [vx300.joint_state_broadcaster_spawner]: Waiting for '/vx300/controller_manager' node to exist
```

lukeschmitt-tr, 2024-09-13:
Did you rebuild your workspace after making the change? Note that doing a symlink'ed installation would work too.

Can you launch gzclient and gzserver manually and individually?

```bash
gzserver --verbose
```

```bash
gzclient --verbose
```

When running xsarm_gz_classic, do you see the arm appear? If so, what state is it in?

YumTaha, 2024-09-13:
I just rebuilt the workspace using ```colcon build --symlink-install```
Launching ```gzclient``` and ```gzserver``` was successful, and there were no error logs in the output. 

I'm encountering the same issue when using commands.
in Gazebo, I only see the walls and not the arm in the middle. In VRim, the arm appears in the home state, but the actual robot's lights are red, indicating it's not connected. 
```ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=vx300 hardware_type:=gz_classic verbose:=true```

Logs
```
pibot@pibot:~/interbotix_ws$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=vx300 hardware_type:=gz_classic verbose:=true
[INFO] [launch]: All log files can be found below /home/pibot/.ros/log/2024-09-13-15-27-48-186106-pibot-20934
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [20947]
[INFO] [rviz2-2]: process started with pid [20949]
[INFO] [gzserver-3]: process started with pid [20951]
[INFO] [gzclient --verbose  -4]: process started with pid [20953]
[INFO] [spawn_entity.py-5]: process started with pid [20957]
[INFO] [robot_state_publisher-6]: process started with pid [20959]
[move_group-1] [WARN] [1726255677.983363524] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1726255677.983833557] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1726255678.039182068] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1726255678.039377741] [moveit_robot_model.robot_model]: Loading robot model 'vx300'...
[move_group-1] [INFO] [1726255678.039486763] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] Link vx300/base_link had 1 children
[move_group-1] Link vx300/shoulder_link had 1 children
[move_group-1] Link vx300/upper_arm_link had 1 children
[move_group-1] Link vx300/forearm_link had 1 children
[move_group-1] Link vx300/wrist_link had 1 children
[move_group-1] Link vx300/gripper_link had 1 children
[move_group-1] Link vx300/ee_arm_link had 2 children
[move_group-1] Link vx300/gripper_prop_link had 0 children
[move_group-1] Link vx300/gripper_bar_link had 1 children
[move_group-1] Link vx300/fingers_link had 3 children
[move_group-1] Link vx300/ee_gripper_link had 0 children
[move_group-1] Link vx300/left_finger_link had 0 children
[move_group-1] Link vx300/right_finger_link had 0 children
[move_group-1] [INFO] [1726255679.355369247] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1726255679.356135344] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/vx300/joint_states' for joint states
[move_group-1] [INFO] [1726255679.362947385] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/vx300/joint_states'
[move_group-1] [INFO] [1726255679.366557903] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1726255679.366682518] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1726255679.371034781] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1726255679.371133487] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1726255679.375753665] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1726255679.388882437] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1726255679.395705571] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1726255679.395876965] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1726255679.395946430] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[gzclient --verbose  -4] Gazebo multi-robot simulator, version 11.11.0
[gzclient --verbose  -4] Copyright (C) 2012 Open Source Robotics Foundation.
[gzclient --verbose  -4] Released under the Apache 2 License.
[gzclient --verbose  -4] http://gazebosim.org
[gzclient --verbose  -4] 
[move_group-1] [INFO] [1726255679.609969024] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1726255679.908541515] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1726255679.946863352] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1726255679.947016301] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[... cut ...]
[gzclient --verbose  -4] 
[gzclient --verbose  -4] 
[ERROR] [gzserver-3]: process has died [pid 20951, exit code -2, cmd 'gzserver /home/pibot/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world      --verbose                                                                -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[gzserver-3] 
[gzserver-3] 
[INFO] [rviz2-2]: process has finished cleanly [pid 20949]
```

YumTaha, 2024-09-13:
I believe I have something not setup with gazebo because even this command ```ros2 launch interbotix_xsarm_sim xsarm_gz_classic.launch.py robot_model:=vx300``` won't run

```
pibot@pibot:~/interbotix_ws$ ros2 launch interbotix_xsarm_sim xsarm_gz_classic.launch.py robot_model:=vx300
[INFO] [launch]: All log files can be found below /home/pibot/.ros/log/2024-09-13-15-44-00-843434-pibot-22407
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [22424]
[INFO] [gzclient   -2]: process started with pid [22426]
[INFO] [spawn_entity.py-3]: process started with pid [22429]
[INFO] [robot_state_publisher-4]: process started with pid [22432]
[INFO] [rviz2-5]: process started with pid [22434]
[spawn_entity.py-3] [INFO] [1726256649.566887260] [spawn_vx300]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1726256649.572781473] [spawn_vx300]: Loading entity published on topic vx300/robot_description
[spawn_entity.py-3] [INFO] [1726256649.592221467] [spawn_vx300]: Waiting for entity xml on vx300/robot_description
[spawn_entity.py-3] [INFO] [1726256649.605090784] [spawn_vx300]: Waiting for service /spawn_entity, timeout = 5
[spawn_entity.py-3] [INFO] [1726256649.608524049] [spawn_vx300]: Waiting for service /spawn_entity
[spawn_entity.py-3] [ERROR] [1726256654.659454760] [spawn_vx300]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-3] [ERROR] [1726256654.674319505] [spawn_vx300]: Spawn service failed. Exiting.
[spawn_entity.py-3] /opt/ros/galactic/lib/python3.8/site-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-3]   warnings.warn(
[ERROR] [spawn_entity.py-3]: process has died [pid 22429, exit code 1, cmd '/opt/ros/galactic/lib/gazebo_ros/spawn_entity.py -entity robot_description -topic vx300/robot_description -x 0.0 -y 0.0 -z 0.0 -Y 0.0 --ros-args -r __node:=spawn_vx300'].
[INFO] [spawner-6]: process started with pid [22528]
[spawner-6] [INFO] [1726256657.529768524] [vx300.joint_state_broadcaster_spawner]: Waiting for '/vx300/controller_manager' node to exist
[spawner-6] [INFO] [1726256659.575286388] [vx300.joint_state_broadcaster_spawner]: Waiting for '/vx300/controller_manager' node to exist
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[spawner-6] Traceback (most recent call last):
[spawner-6]   File "/opt/ros/galactic/lib/controller_manager/spawner", line 20, in <module>
[spawner-6]     ret = main()
[spawner-6]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 162, in main
[spawner-6]     if not wait_for_controller_manager(node, controller_manager_name,
[spawner-6]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 95, in wait_for_controller_manager
[spawner-6]     node_and_namespace = wait_for_value_or(
[spawner-6]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 57, in wait_for_value_or
[spawner-6]     time.sleep(0.2)
[spawner-6] KeyboardInterrupt
[ERROR] [gzclient   -2]: process has died [pid 22426, exit code -2, cmd 'gzclient   '].
[INFO] [robot_state_publisher-4]: process has finished cleanly [pid 22432]
[ERROR] [spawner-6]: process has died [pid 22528, exit code -2, cmd '/opt/ros/galactic/lib/controller_manager/spawner -c vx300/controller_manager joint_state_broadcaster --ros-args -r __node:=joint_state_broadcaster_spawner -r __ns:=/vx300 --params-file /tmp/launch_params_e31stwdj'].
[ERROR] [gzserver-1]: process has died [pid 22424, exit code -2, cmd 'gzserver /home/pibot/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world                                                                      -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[INFO] [rviz2-5]: process has finished cleanly [pid 22434]
pibot@pibot:~/interbotix_ws$ 
```

this is the output with ```verbose:=true```
```
pibot@pibot:~/interbotix_ws$ ros2 launch interbotix_xsarm_sim xsarm_gz_classic.launch.py robot_model:=vx300 verbose:=true
[INFO] [launch]: All log files can be found below /home/pibot/.ros/log/2024-09-13-15-45-44-856382-pibot-22608
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [22620]
[INFO] [gzclient --verbose  -2]: process started with pid [22622]
[INFO] [spawn_entity.py-3]: process started with pid [22625]
[INFO] [robot_state_publisher-4]: process started with pid [22628]
[INFO] [rviz2-5]: process started with pid [22630]
[gzclient --verbose  -2] Gazebo multi-robot simulator, version 11.11.0
[gzclient --verbose  -2] Copyright (C) 2012 Open Source Robotics Foundation.
[gzclient --verbose  -2] Released under the Apache 2 License.
[gzclient --verbose  -2] http://gazebosim.org
[gzclient --verbose  -2] 
[spawn_entity.py-3] [INFO] [1726256755.134109487] [spawn_vx300]: Spawn Entity started
[spawn_entity.py-3] [INFO] [1726256755.137343489] [spawn_vx300]: Loading entity published on topic vx300/robot_description
[spawn_entity.py-3] [INFO] [1726256755.149719460] [spawn_vx300]: Waiting for entity xml on vx300/robot_description
[gzserver-1] Gazebo multi-robot simulator, version 11.11.0
[gzserver-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gzserver-1] Released under the Apache 2 License.
[gzserver-1] http://gazebosim.org
[gzserver-1] 
[... cut ...]
[gzserver-1] [Msg] Loading world file [/home/pibot/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world]
[ERROR] [gzserver-1]: process has died [pid 22620, exit code -2, cmd 'gzserver /home/pibot/interbotix_ws/install/interbotix_common_sim/share/interbotix_common_sim/worlds/interbotix.world      --verbose                                                                -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
[gzserver-1] 
[gzserver-1] 
pibot@pibot:~/interbotix_ws$ 
```

lukeschmitt-tr, 2024-09-13:
See if you have the same issues when running the standard spawn entity demo from the gazebo_ros package:

```
ros2 launch gazebo_ros spawn_entity_demo.launch.py
```

YumTaha, 2024-09-13:
failed. do I need more arguments or what?
```
pibot@pibot:~/interbotix_ws$ ros2 launch gazebo_ros spawn_entity_demo.launch.py
[INFO] [launch]: All log files can be found below /home/pibot/.ros/log/2024-09-13-15-51-38-550422-pibot-22962
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/opt/ros/galactic/share/gazebo_ros/launch/spawn_entity_demo.launch.py", line 34, in generate_launch_description
    spawn_entity = Node(package='gazebo_ros', node_executable='spawn_entity.py',
TypeError: __init__() missing 1 required keyword-only argument: 'executable'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: __init__() missing 1 required keyword-only argument: 'executable'
pibot@pibot:~/interbotix_ws$ 
```

closed 2024-10-31 by YumTaha