Interbotix/interbotix_ros_manipulators #221: [Bug]: MoveIt is not connecting to actual robot

YumTaha, 2024-09-12:
### What happened?

When I execute the command ```ros2 launch interbotix_xsarm_moveit_interface xsarm_moveit_interface.launch.py robot_model:=vx300```, everything runs smoothly up until the end. However, the xs_sdk node unexpectedly crashes, causing the robot to disconnect and become unresponsive, leading to failed planned movements.

### Robot Model

vx300

### Operating System

Ubuntu 20.04

### ROS Distro

ROS 2 Galactic

### Relevant log output

```shell
pibot@pibot:~$ ros2 launch interbotix_xsarm_moveit_interface xsarm_moveit_interface.launch.py robot_model:=vx300
[INFO] [launch]: All log files can be found below /home/pibot/.ros/log/2024-09-12-13-00-58-747759-pibot-2400
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [moveit_interface-1]: process started with pid [2411]
[INFO] [moveit_interface_gui-2]: process started with pid [2413]
[INFO] [move_group-3]: process started with pid [2415]
[INFO] [rviz2-4]: process started with pid [2417]
[INFO] [ros2_control_node-5]: process started with pid [2419]
[INFO] [spawner-6]: process started with pid [2421]
[INFO] [spawner-7]: process started with pid [2423]
[INFO] [xs_sdk-8]: process started with pid [2425]
[INFO] [robot_state_publisher-9]: process started with pid [2427]
[xs_sdk-8] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.3'.
[xs_sdk-8] [INFO] Using logging level 'INFO'.
[xs_sdk-8] [INFO] Loaded mode configs from '/home/pibot/interbotix_ws/install/interbotix_xsarm_moveit_interface/share/interbotix_xsarm_moveit_interface/config/modes.yaml'.
[xs_sdk-8] [INFO] Loaded motor configs from '/home/pibot/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/vx300.yaml'.
[xs_sdk-8] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  7, Model: 'XM430-W350', Joint Name: 'wrist_rotate'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  6, Model: 'XM540-W270', Joint Name: 'wrist_angle'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  5, Model: 'XM540-W270', Joint Name: 'elbow_shadow'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  4, Model: 'XM540-W270', Joint Name: 'elbow'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XM540-W270', Joint Name: 'shoulder'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  8, Model: 'XM430-W350', Joint Name: 'gripper'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XM540-W270', Joint Name: 'shoulder_shadow'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XM540-W270', Joint Name: 'waist'.
[spawner-6] [INFO] [1726146065.737277978] [vx300.arm_controller_spawner]: Waiting for '/vx300/controller_manager' services to be available
[xs_sdk-8] [WARN] Writing startup register values to EEPROM. This only needs to be done once on a robot if using a default motor config file, or after a motor config file has been modified. Can set `write_eeprom_on_startup` to false from now on.
[move_group-3] [WARN] [1726146065.968102442] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-3] [WARN] [1726146065.970286302] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[spawner-7] [INFO] [1726146065.981300675] [vx300.gripper_controller_spawner]: Waiting for '/vx300/controller_manager' services to be available
[move_group-3] [INFO] [1726146065.992577676] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0204921 seconds
[move_group-3] [INFO] [1726146065.992783157] [moveit_robot_model.robot_model]: Loading robot model 'vx300'...
[move_group-3] [INFO] [1726146065.992829416] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[xs_sdk-8] [INFO] The operating mode for the 'arm' group was changed to 'position' with profile type 'velocity'.
[xs_sdk-8] [INFO] The operating mode for the 'gripper' joint was changed to 'linear_position' with profile type 'velocity'.
[xs_sdk-8] [INFO] Interbotix X-Series Driver is up!
[move_group-3] Link vx300/base_link had 1 children
[move_group-3] Link vx300/shoulder_link had 1 children
[move_group-3] Link vx300/upper_arm_link had 1 children
[move_group-3] Link vx300/forearm_link had 1 children
[move_group-3] Link vx300/wrist_link had 1 children
[move_group-3] Link vx300/gripper_link had 1 children
[move_group-3] Link vx300/ee_arm_link had 2 children
[move_group-3] Link vx300/gripper_prop_link had 0 children
[move_group-3] Link vx300/gripper_bar_link had 1 children
[move_group-3] Link vx300/fingers_link had 3 children
[move_group-3] Link vx300/ee_gripper_link had 0 children
[move_group-3] Link vx300/left_finger_link had 0 children
[move_group-3] Link vx300/right_finger_link had 0 children
[ros2_control_node-5] [INFO] [1726146066.546050094] [vx300.xs_hardware_interface]: Joint states received.
[ros2_control_node-5] [INFO] [1726146066.614235753] [vx300.controller_manager]: update rate is 10 Hz
[... 70 lines cut ...]
[move_group-3] Loading 'move_group/ClearOctomapService'...
[move_group-3] Loading 'move_group/MoveGroupCartesianPathService'...
[move_group-3] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-3] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[move_group-3] Loading 'move_group/MoveGroupKinematicsService'...
[move_group-3] Loading 'move_group/MoveGroupMoveAction'...
[move_group-3] Loading 'move_group/MoveGroupPlanService'...
[move_group-3] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-3] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-3] 
[move_group-3] You can start planning now!
[move_group-3] 
[moveit_interface-1] [INFO] [1726146069.258648272] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.305965 seconds
[moveit_interface-1] [INFO] [1726146069.258975882] [moveit_robot_model.robot_model]: Loading robot model 'vx300'...
[moveit_interface-1] [INFO] [1726146069.259065604] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[moveit_interface-1] [WARN] [1726146069.498752802] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[moveit_interface-1] [INFO] [1726146069.605428174] [move_group_interface]: Ready to take commands for planning group interbotix_arm.
[moveit_interface-1] [INFO] [1726146069.609353249] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[moveit_interface-1] [INFO] [1726146069.646635655] [interbotix_moveit_node.rviz_visual_tools]: Topic /moveit_visual_tools waiting 5 seconds for subscriber, 
[moveit_interface-1] [WARN] [1726146074.648094849] [interbotix_moveit_node.rviz_visual_tools]: Topic /moveit_visual_tools unable to connect to any subscribers within 5 sec. It is possible initially published visual messages will be lost.
[moveit_interface-1] [INFO] [1726146074.648497070] [interbotix_moveit_node]: Reference frame: world
[moveit_interface-1] [INFO] [1726146074.649052141] [interbotix_moveit_node]: End effector link: vx300/ee_gripper_link
[xs_sdk-8] *** stack smashing detected ***: terminated
[ERROR] [xs_sdk-8]: process has died [pid 2425, exit code -6, cmd '/home/pibot/interbotix_ws/install/interbotix_xs_sdk/lib/interbotix_xs_sdk/xs_sdk --ros-args -r __node:=xs_sdk -r __ns:=/vx300 --params-file /tmp/launch_params_iwph0kab'].
```

lukeschmitt-tr, 2024-09-12:
> ```
> [xs_sdk-8] *** stack smashing detected ***: terminated
> ```

We sometimes see this issue when there are multiple xs_sdk nodes being launched simultaneously or some other issue related to port ownership. Please ensure that you are only loaded a single xs_sdk node at a time. Note that the xsarm_moveit launch file already launches the node for you so you do not have to launch it yourself in another terminal.

YumTaha, 2024-09-13:
I usually run my commands directly in the terminal because I'm still learning how to use the robot. Currently, when I execute a command, it connects to the robot but only executes the first plan. Subsequent attempts to execute other plans fail. 
```
moveit_interface-1] [INFO] [1726232345.129946999] [move_group_interface]: Planning request complete!
[moveit_interface-1] [INFO] [1726232345.230586705] [move_group_interface]: time taken to generate plan: 3.02074 seconds
[moveit_interface-1] [INFO] [1726232345.234172376] [interbotix_moveit_node]: Plan success: 1
[moveit_interface_gui-2] [INFO] [1726232345.257625620] [moveit_interface_gui]: Planning EE position was successful!
[move_group-3] [INFO] [1726232348.560617978] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-3] [INFO] [1726232348.561425743] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-3] [INFO] [1726232348.561641170] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-3] [INFO] [1726232348.562040784] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[moveit_interface-1] [INFO] [1726232348.764842898] [move_group_interface]: Execute request accepted
[move_group-3] [INFO] [1726232349.562443561] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[move_group-3] Check clock synchronization if your are running ROS across multiple machines!
[move_group-3] [WARN] [1726232349.562592062] [moveit_ros.trajectory_execution_manager]: Failed to validate trajectory: couldn't receive full current joint state within 1s
[move_group-3] [INFO] [1726232349.562719340] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[moveit_interface-1] [INFO] [1726232349.873638922] [move_group_interface]: Execute request aborted
[moveit_interface-1] [ERROR] [1726232349.974110983] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
[moveit_interface_gui-2] [INFO] [1726232349.984315609] [moveit_interface_gui]: Execution was not successful.
```

lukeschmitt-tr, 2024-09-13:
Please provide the full logs. You may also try setting the xs_driver_logging_level launch arg to DEBUG to get more information out of the xs_sdk and xs_driver.

YumTaha, 2024-09-13:
Here is the complete log from two attempts using the standard log settings. The third log has XS logging configured to DEBUG mode.

**1st LOG**
```
pibot@pibot:~/interbotix_ws$ ros2 launch interbotix_xsarm_moveit_interface xsarm_moveit_interface.launch.py robot_model:=vx300
[INFO] [launch]: All log files can be found below /home/pibot/.ros/log/2024-09-13-11-38-54-731877-pibot-5020
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [moveit_interface-1]: process started with pid [5029]
[INFO] [moveit_interface_gui-2]: process started with pid [5031]
[INFO] [move_group-3]: process started with pid [5033]
[INFO] [rviz2-4]: process started with pid [5035]
[INFO] [ros2_control_node-5]: process started with pid [5037]
[INFO] [spawner-6]: process started with pid [5039]
[INFO] [spawner-7]: process started with pid [5041]
[INFO] [xs_sdk-8]: process started with pid [5043]
[INFO] [robot_state_publisher-9]: process started with pid [5045]
[xs_sdk-8] [INFO] Using Interbotix X-Series Driver Version: 'v0.3.3'.
[xs_sdk-8] [INFO] Using logging level 'INFO'.
[xs_sdk-8] [INFO] Loaded mode configs from '/home/pibot/interbotix_ws/install/interbotix_xsarm_moveit_interface/share/interbotix_xsarm_moveit_interface/config/modes.yaml'.
[xs_sdk-8] [INFO] Loaded motor configs from '/home/pibot/interbotix_ws/install/interbotix_xsarm_control/share/interbotix_xsarm_control/config/vx300.yaml'.
[xs_sdk-8] [INFO] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  7, Model: 'XM430-W350', Joint Name: 'wrist_rotate'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  6, Model: 'XM540-W270', Joint Name: 'wrist_angle'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  5, Model: 'XM540-W270', Joint Name: 'elbow_shadow'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  4, Model: 'XM540-W270', Joint Name: 'elbow'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  2, Model: 'XM540-W270', Joint Name: 'shoulder'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  8, Model: 'XM430-W350', Joint Name: 'gripper'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  3, Model: 'XM540-W270', Joint Name: 'shoulder_shadow'.
[xs_sdk-8] [INFO] 	Found DYNAMIXEL ID:  1, Model: 'XM540-W270', Joint Name: 'waist'.
[move_group-3] [WARN] [1726241941.429319557] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-3] [WARN] [1726241941.429567388] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-3] [INFO] [1726241941.455688674] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0244237 seconds
[move_group-3] [INFO] [1726241941.456002004] [moveit_robot_model.robot_model]: Loading robot model 'vx300'...
[move_group-3] [INFO] [1726241941.456096021] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-3] Link vx300/base_link had 1 children
[move_group-3] Link vx300/shoulder_link had 1 children
[move_group-3] Link vx300/upper_arm_link had 1 children
[move_group-3] Link vx300/forearm_link had 1 children
[move_group-3] Link vx300/wrist_link had 1 children
[move_group-3] Link vx300/gripper_link had 1 children
[move_group-3] Link vx300/ee_arm_link had 2 children
[move_group-3] Link vx300/gripper_prop_link had 0 children
[move_group-3] Link vx300/gripper_bar_link had 1 children
[move_group-3] Link vx300/fingers_link had 3 children
[move_group-3] Link vx300/ee_gripper_link had 0 children
[move_group-3] Link vx300/left_finger_link had 0 children
[move_group-3] Link vx300/right_finger_link had 0 children
[xs_sdk-8] [WARN] Writing startup register values to EEPROM. This only needs to be done once on a robot if using a default motor config file, or after a motor config file has been modified. Can set `write_eeprom_on_startup` to false from now on.
[xs_sdk-8] [INFO] The operating mode for the 'arm' group was changed to 'position' with profile type 'velocity'.
[xs_sdk-8] [INFO] The operating mode for the 'gripper' joint was changed to 'linear_position' with profile type 'velocity'.
[xs_sdk-8] [INFO] Interbotix X-Series Driver is up!
[move_group-3] [INFO] [1726241942.381023143] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-3] [INFO] [1726241942.381831023] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/vx300/joint_states' for joint states
[move_group-3] [INFO] [1726241942.389103128] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/vx300/joint_states'
[move_group-3] [INFO] [1726241942.409314850] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-3] [INFO] [1726241942.410664872] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-3] [INFO] [1726241942.418634006] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-3] [INFO] [1726241942.418761468] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-3] [INFO] [1726241942.428196234] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[... cut ...]
[xs_sdk-8] [DEBUG] ID: 8, writing linear_position command nan.
[xs_sdk-8] [DEBUG] ID: 8, writing linear_position command nan.
[xs_sdk-8] [DEBUG] ID: 8, writing linear_position command nan.
[xs_sdk-8] [DEBUG] ID: 8, writing linear_position command nan.
[xs_sdk-8] *** stack smashing detected ***: terminated
[ERROR] [xs_sdk-8]: process has died [pid 6506, exit code -6, cmd '/home/pibot/interbotix_ws/install/interbotix_xs_sdk/lib/interbotix_xs_sdk/xs_sdk --ros-args -r __node:=xs_sdk -r __ns:=/vx300 --params-file /tmp/launch_params_h87hno18'].
```

closed 2024-10-31 by YumTaha