Interbotix/interbotix_ros_manipulators #227: [Bug]: Gripper Malfunction on Robotic Arm WX200

anugraha05, 2024-10-09:
### What happened?

We are experiencing an intermittent issue with the gripper on the wx200 robotic arm. During the release action, the gripper sometimes extends to its maximum position and then unexpectedly retracts halfway before stopping. This behaviour occurs without any corresponding code logic, and there is no discernible pattern to when it happens. The release and grasp functions defined in the package are independent of the position of the gripper and is only dependent on the effort command which makes it difficult to manipulate left and right gripper positions. 

### Robot Model

wx200

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Humble

### Steps To Reproduce

Issue the release command to the gripper.
Observe the gripper's movement (intermittent behavior, may not always occur).

lukeschmitt-tr, 2024-10-09:
Please share the commit hashes for your interbotix_ros_core, interbotix_ros_manipulators, and interbotix_ros_toolboxes repos.

anugraha05, 2024-10-10:
I ran the gripper_control.py demo file in " interbotix_ros_xsarms/examples/python_demos/gripper_control.py " for testing the release and grasp. The issue occurs randomly. PFA the py code used for testing,
```
from interbotix_common_modules.common_robot.robot import robot_shutdown, robot_startup
 
from interbotix_xs_modules.xs_robot.arm import InterbotixManipulatorXS
 
def main():

    bot = InterbotixManipulatorXS(
        robot_model='wx200',
        group_name='arm',
        gripper_name='gripper',
    )

    robot_startup()

    bot.gripper.grasp(2.0)
    bot.gripper.release(2.0)
    bot.gripper.set_pressure(1.0)
    bot.gripper.grasp(2.0)
    bot.gripper.release(2.0)

    robot_shutdown()

if __name__ == '__main__':
    main()
```