Interbotix/interbotix_ros_manipulators #228: [Bug]: No robot spawning in Rviz when running MoveIt 2 Motion Planning

SwooshMihu, 2024-10-09:
### What happened?

When running 
`ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=rx200 hardware_type:=fake`
Rviz shows up but the robot arm does not load. 
Running `ros2 launch interbotix_xsarm_control xsarm_control.launch.py robot_model:=rx200 use_sim:=true` works just fine.

### Robot Model

rx200

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Humble

### Relevant log output

```shell
[INFO] [launch]: All log files can be found below /home/mihneas/.ros/log/2024-10-09-17-04-41-520891-mihnea-dell-8931
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [8935]
[INFO] [rviz2-2]: process started with pid [8937]
[INFO] [ros2_control_node-3]: process started with pid [8939]
[INFO] [spawner-4]: process started with pid [8941]
[INFO] [spawner-5]: process started with pid [8943]
[INFO] [spawner-6]: process started with pid [8945]
[INFO] [robot_state_publisher-7]: process started with pid [8947]
[ros2_control_node-3] [WARN] [1728482682.198968373] [rx200.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1728482682.199186382] [resource_manager]: Loading hardware 'XSHardwareInterface' 
[ros2_control_node-3] [INFO] [1728482682.199464955] [resource_manager]: Initialize hardware 'XSHardwareInterface' 
[ros2_control_node-3] [WARN] [1728482682.199479415] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-3] <state_interface name="velocity"> 
[ros2_control_node-3]   <param name="initial_value">0.0</param> 
[ros2_control_node-3] </state_interface>
[ros2_control_node-3] [INFO] [1728482682.199485942] [resource_manager]: Successful initialization of hardware 'XSHardwareInterface'
[ros2_control_node-3] [INFO] [1728482682.199517824] [resource_manager]: 'configure' hardware 'XSHardwareInterface' 
[ros2_control_node-3] [INFO] [1728482682.199521785] [resource_manager]: Successful 'configure' of hardware 'XSHardwareInterface'
[ros2_control_node-3] [INFO] [1728482682.199524857] [resource_manager]: 'activate' hardware 'XSHardwareInterface' 
[ros2_control_node-3] [INFO] [1728482682.199527573] [resource_manager]: Successful 'activate' of hardware 'XSHardwareInterface'
[ros2_control_node-3] [INFO] [1728482682.201265459] [rx200.controller_manager]: update rate is 10 Hz
[ros2_control_node-3] [WARN] [1728482682.201376681] [rx200.controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[move_group-1] [WARN] [1728482682.215460475] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1728482682.218008595] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00239452 seconds
[move_group-1] [INFO] [1728482682.218047941] [moveit_robot_model.robot_model]: Loading robot model 'rx200'...
[move_group-1] [INFO] [1728482682.218060864] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [INFO] [1728482682.276129109] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1728482682.276241384] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/rx200/joint_states' for joint states
[move_group-1] [INFO] [1728482682.276581361] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/rx200/joint_states'
[move_group-1] [INFO] [1728482682.276732208] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1728482682.276741135] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1728482682.276828995] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1728482682.276834610] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1728482682.276933610] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1728482682.277037784] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1728482682.277232041] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1728482682.277244147] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1728482682.277250667] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1728482682.280309832] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1728482682.286810649] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1728482682.288134320] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1728482682.288143999] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1728482682.288147016] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1728482682.288155204] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1728482682.288163555] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[move_group-1] [INFO] [1728482682.288169642] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1728482682.288175883] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[... 52 lines cut ...]
[ros2_control_node-3] [WARN] [1728482682.615607706] [rx200.arm_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[ros2_control_node-3] [INFO] [1728482682.702555911] [rx200.controller_manager]: Configuring controller 'gripper_controller'
[ros2_control_node-3] [INFO] [1728482682.703080612] [rx200.gripper_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-3] [INFO] [1728482682.703199710] [rx200.gripper_controller]: Command interfaces are [position] and state interfaces are [position].
[ros2_control_node-3] [INFO] [1728482682.703280641] [rx200.gripper_controller]: Using 'splines' interpolation method.
[ros2_control_node-3] [INFO] [1728482682.705807423] [rx200.gripper_controller]: Controller state will be published at 50.00 Hz.
[ros2_control_node-3] [INFO] [1728482682.708851813] [rx200.gripper_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-4] [INFO] [1728482682.723282298] [rx200.arm_controller_spawner]: Loaded arm_controller
[ros2_control_node-3] [INFO] [1728482682.802456270] [rx200.controller_manager]: Configuring controller 'arm_controller'
[ros2_control_node-3] [INFO] [1728482682.802912373] [rx200.arm_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-3] [INFO] [1728482682.802954762] [rx200.arm_controller]: Command interfaces are [position] and state interfaces are [position].
[ros2_control_node-3] [INFO] [1728482682.803024771] [rx200.arm_controller]: Using 'splines' interpolation method.
[ros2_control_node-3] [INFO] [1728482682.804115472] [rx200.arm_controller]: Controller state will be published at 50.00 Hz.
[... cut ...]
[INFO] [spawner-6]: process has finished cleanly [pid 8945]
[spawner-5] [INFO] [1728482683.304676211] [rx200.gripper_controller_spawner]: Configured and activated gripper_controller
[INFO] [spawner-5]: process has finished cleanly [pid 8943]
[spawner-4] [INFO] [1728482683.504488279] [rx200.arm_controller_spawner]: Configured and activated arm_controller
[INFO] [spawner-4]: process has finished cleanly [pid 8941]
```

### Additional Info

Another thing that happens right now is that gzclient doesen't work either, crashing immediately.

lukeschmitt-tr, 2024-10-09:
Based on your terminal output, it seems like everything spawned in correctly. Please provide the following:
* Commit hashes for your interbotix_ros_core, ros_manipulators, and ros_toolboxes repos
* A screenshot of RViz

SwooshMihu, 2024-10-09:
[image]

interbotix_ros_core hash: 
```
commit f0ae683d1bd1fa6d65983e358765ec53ffb3002e (HEAD -> humble, origin/humble)
Author: lukeschmitt-tr <85308904+lukeschmitt-tr@users.noreply.github.com>
Date:   Thu Aug 22 12:26:41 2024 -0500

    Update X-Series driver to v0.3.7 (#54)
```

ros_toolboxes hash:
```
commit daae95ca3a5952e3e1d993e4900ab40c3ede8ec8 (HEAD -> humble, origin/humble)
Author: lukeschmitt-tr <85308904+lukeschmitt-tr@users.noreply.github.com>
Date:   Tue Jun 25 19:49:38 2024 -0500

    Add default to FootswitchTrigger trigger (#74)
```
ros_manipulators hash:
```
commit 3eecabf18fc4e097f2ed3da230c543416e25b359 (HEAD -> humble, origin/humble)
Author: Luke Schmitt <luke@trossenrobotics.com>
Date:   Tue Sep 3 19:01:36 2024 +0000

    Fix xsarm_puppet package dependencies
```

lukeschmitt-tr, 2024-10-11:
I am unable to replicate this issue and it seems like everything has been brought up correctly on your end.

Please make sure that the following options are selected in the MotionPlanning display:
* Robot Description: `robot_description`
* Planning Scene Topic: `/monitored_planning_scene`
* Planning Request > Planning Group: `interbotix_arm`
* Planning Request > Query Goal State: ✅ 

[image]

```shell
$ ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=rx200 hardware_type:=fake
[INFO] [launch]: All log files can be found below /home/interbotix/.ros/log/2024-10-11-15-48-30-292317-desktop-385520
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [385530]
[INFO] [rviz2-2]: process started with pid [385532]
[INFO] [ros2_control_node-3]: process started with pid [385534]
[INFO] [spawner-4]: process started with pid [385536]
[INFO] [spawner-5]: process started with pid [385538]
[INFO] [spawner-6]: process started with pid [385540]
[INFO] [robot_state_publisher-7]: process started with pid [385542]
[move_group-1] [WARN] [1728661710.797480385] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[ros2_control_node-3] [WARN] [1728661710.807155897] [rx200.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1728661710.807687223] [resource_manager]: Loading hardware 'XSHardwareInterface' 
[ros2_control_node-3] [INFO] [1728661710.808690036] [resource_manager]: Initialize hardware 'XSHardwareInterface' 
[ros2_control_node-3] [WARN] [1728661710.808909525] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-3] <state_interface name="velocity"> 
[ros2_control_node-3]   <param name="initial_value">0.0</param> 
[ros2_control_node-3] </state_interface>
[ros2_control_node-3] [INFO] [1728661710.808950432] [resource_manager]: Successful initialization of hardware 'XSHardwareInterface'
[ros2_control_node-3] [INFO] [1728661710.809408972] [resource_manager]: 'configure' hardware 'XSHardwareInterface' 
[ros2_control_node-3] [INFO] [1728661710.809425359] [resource_manager]: Successful 'configure' of hardware 'XSHardwareInterface'
[ros2_control_node-3] [INFO] [1728661710.809437833] [resource_manager]: 'activate' hardware 'XSHardwareInterface' 
[ros2_control_node-3] [INFO] [1728661710.809442218] [resource_manager]: Successful 'activate' of hardware 'XSHardwareInterface'
[ros2_control_node-3] [INFO] [1728661710.814662578] [rx200.controller_manager]: CREATED CONTROLLER MANAGER!!!!
[ros2_control_node-3] [INFO] [1728661710.814694999] [rx200.controller_manager]: update rate is 10 Hz
[ros2_control_node-3] [INFO] [1728661710.814829039] [rx200.controller_manager]: RT kernel is recommended for better performance
[move_group-1] [INFO] [1728661710.826392169] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.028791 seconds
[move_group-1] [INFO] [1728661710.826421054] [moveit_robot_model.robot_model]: Loading robot model 'rx200'...
[move_group-1] [INFO] [1728661710.826426547] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [INFO] [1728661710.871923634] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1728661710.872026018] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/rx200/joint_states' for joint states
[move_group-1] [INFO] [1728661710.872388624] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/rx200/joint_states'
[move_group-1] [INFO] [1728661710.872574923] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1728661710.872585837] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1728661710.872689059] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1728661710.872697119] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1728661710.872817908] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1728661710.872978903] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1728661710.873491020] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1728661710.873503751] [moveit.ros.occupancy_map_monitor.middleware_handle]: No sensor plugin specified for octomap updater ; ignoring.
[move_group-1] [INFO] [1728661710.873511750] [moveit.ros.occupancy_map_monitor.middleware_handle]: Skipping octomap updater plugin ''
[move_group-1] [INFO] [1728661710.877191088] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[... cut ...]
[ros2_control_node-3] [INFO] [1728661711.816120214] [rx200.arm_controller]: Controller state will be published at 50.00 Hz.
[ros2_control_node-3] [INFO] [1728661711.816561388] [rx200.arm_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-5] [INFO] [1728661712.116161372] [rx200.gripper_controller_spawner]: Configured and activated gripper_controller
[INFO] [spawner-5]: process has finished cleanly [pid 385538]
[spawner-4] [INFO] [1728661712.315847009] [rx200.arm_controller_spawner]: Configured and activated arm_controller
[INFO] [spawner-4]: process has finished cleanly [pid 385536]
```

SwooshMihu, 2024-10-14:
[image]
From what i can notice, the _Status: OK_ and _Planning Scene : Planning Scene Loaded Succesfully_ are missing from my Motion Planning display.

SwooshMihu, 2024-10-14:
[image]
Also, it seems there was no motion planning library loaded. ( This is visible in the Context panel of the MotionPlanning GUI).

closed 2024-11-20 by SwooshMihu