Interbotix/interbotix_ros_manipulators #231: [Bug]: nothing is published to topic:    /pc_filter/markers/crop_box

YumTaha, 2024-10-31:
### What happened?

 Initial setup after installing ROS 2 Humble and the Interbotix workspace. Installed the RealSense package using the following command: `sudo apt install ros-humble-realsense2-camera`
 When running perception, RViz did not display the camera (no camera)(mind you the `realsense-viewer` works fine). It indicated an issue with the topic `/camera/color/image_raw`. After executing `ros2 topic list`, I discovered that the correct topic was: `/camera/camera/color/image_raw`.
```
iras@pibot:~/interbotix_ws$ ros2 topic list
/apriltag/tag_detections
/camera/camera/color/camera_info
/camera/camera/color/image_raw
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/color/points
/camera/camera/depth/image_rect_raw
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_color
/camera/depth/color/points
/clicked_point
/goal_pose
/initialpose
/parameter_events
/pc_filter/markers/crop_box
/pc_filter/markers/crop_box_array
/pc_filter/markers/objects
/pc_filter/markers/objects_array
/pc_filter/pointcloud/filtered
/pc_filter/pointcloud/objects
/rosout
/static_transforms
/tf
/tf_static
/vx300/commands/joint_group
/vx300/commands/joint_single
/vx300/commands/joint_trajectory
/vx300/joint_states
/vx300/robot_description
```

 To verify this, I modified the RViz interface by changing the image topic to: `/camera/camera/color/image_raw` and the image displayed correctly. I then updated the launch configuration and Python files, changing the topics:
 From `/camera/color/image_raw` to `/camera/camera/color/image_raw` and from `/camera/color/camera_info` to `/camera/camera/color/camera_info`. This resolved the initial issue, but now I face another problem. Although the camera captures images of the AprilTag and recognizes its position in relation to the robot arm, **the CropBox is not visible in RViz**. The interface indicates that the status is OK, yet nothing is being published to the topic `/pc_filter/markers/crop_box`.

### Robot Model

vx300

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Humble

### Relevant log output

```shell
iras@pibot:~/interbotix_ws$ ros2 launch interbotix_xsarm_perception xsarm_perception.launch.py robot_model:=vx300 use_pointcloud_tuner_gui:=true use_armtag_tuner_gui:=true
[INFO] [launch]: All log files can be found below /home/iras/.ros/log/2024-10-31-16-26-53-393536-pibot-56575
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-3]: process started with pid [56581]
[INFO] [xs_sdk-1]: process started with pid [56577]
[INFO] [robot_state_publisher-2]: process started with pid [56579]
[INFO] [pointcloud_pipeline-4]: process started with pid [56583]
[INFO] [pointcloud_tuner_gui-5]: process started with pid [56585]
[INFO] [apriltag_ros_single_image_detector_node-6]: process started with pid [56587]
[INFO] [picture_snapper-7]: process started with pid [56589]
[INFO] [armtag_tuner_gui-8]: process started with pid [56591]
[INFO] [static_trans_pub.py-9]: process started with pid [56593]
[INFO] [rviz2-10]: process started with pid [56595]
[xs_sdk-1] [INFO] [1730406414.026200] Using Interbotix X-Series Driver Version: 'v0.3.7'.
[xs_sdk-1] [DEBUG] [1730406414.027184] Set logging level to 0.
[... 98 lines cut ...]
[xs_sdk-1] [DEBUG] [1730406415.336100] ID: 7, torqued off.
[pointcloud_tuner_gui-5] [INFO] [1730406415.346723718] [pc_filter.interbotix_robot_manipulation]: FilterParam filepath: '/home/iras/interbotix_ws/install/interbotix_xsarm_perception/share/interbotix_xsarm_perception/config/filter_params.yaml'
[xs_sdk-1] [DEBUG] [1730406415.348945] ID: 7, write Drive_Mode [00000100].
[xs_sdk-1] [DEBUG] [1730406415.350097] ID: 7, skipping set mode position.
[xs_sdk-1] [DEBUG] [1730406415.372364] ID: 7, set mode position, prof_vel=2000, prof_acc=300.
[xs_sdk-1] [DEBUG] [1730406415.383179] ID: 7, torqued on.
[xs_sdk-1] [INFO] [1730406415.383230] The operating mode for the 'arm' group was changed to 'position' with profile type 'time'.
[xs_sdk-1] [DEBUG] [1730406415.384086] ID: 8, read Drive_Mode [00000000].
[pointcloud_tuner_gui-5] [INFO] [1730406415.385069126] [pc_filter.interbotix_robot_manipulation]: Initialized InterbotixPointCloudInterface.
[xs_sdk-1] [DEBUG] [1730406415.385089] ID: 8, skipping set mode pwm.
[xs_sdk-1] [DEBUG] [1730406415.385108] ID: 8, set mode pwm.
[xs_sdk-1] [DEBUG] [1730406415.396049] ID: 8, torqued on.
[xs_sdk-1] [INFO] [1730406415.396078] The operating mode for the 'gripper' joint was changed to 'pwm' with profile type 'velocity'.
[xs_sdk-1] [INFO] [1730406415.396112] Interbotix X-Series Driver is up!
[picture_snapper-7] [INFO] [1730406415.514099732] [apriltag.picture_snapper]: Saving images to '/tmp'.
[pointcloud_tuner_gui-5] [INFO] [1730406415.585774496] [pc_filter.interbotix_robot_manipulation]: PointCloud Tuner GUI is up!
[xs_sdk-1] [INFO] [1730406415.611121802] [interbotix_xs_sdk.xs_sdk]: InterbotixRobotXS is up!
[armtag_tuner_gui-8] [WARN] [1730406416.207853515] [armtag.interbotix_robot_manipulation]: Service '/apriltag/snap_picture' not yet ready.
[picture_snapper-7] [INFO] [1730406417.745248665] [apriltag.picture_snapper]: Ready to save images from topic 'camera/camera/color/image_raw'.
[picture_snapper-7] [INFO] [1730406417.746031595] [apriltag.picture_snapper]: Picture Snapper is up!
[armtag_tuner_gui-8] [INFO] [1730406418.967949494] [armtag.interbotix_robot_manipulation]: CameraInfo message received! Continuing...
[armtag_tuner_gui-8] [INFO] [1730406418.969230662] [armtag.interbotix_robot_manipulation]: Initialized InterbotixAprilTagInterface!
[armtag_tuner_gui-8] [INFO] [1730406418.974876627] [armtag.interbotix_robot_manipulation]: Initialized InterbotixArmTagInterface!
[armtag_tuner_gui-8] [INFO] [1730406419.116332383] [armtag.interbotix_robot_manipulation]: ArmTag Tuner GUI is up!
```

lukeschmitt-tr, 2024-11-01:
The realsense2_camera package was somewhat recently updated to namespace the camera output topics under the camera name. Please try setting the [`cloud_topic`](https://github.com/Interbotix/interbotix_ros_manipulators/blob/humble/interbotix_ros_xsarms/interbotix_xsarm_perception/launch/xsarm_perception.launch.py#L343-L348) launch arg to `'/camera/camera/depth/color/points'`.

The reason you're not seeing the cropbox (or likely any other markers) is because those visualizations are typically only published on receiving a new pointcloud message. See the [pointcloud pipeline source](https://github.com/Interbotix/interbotix_ros_toolboxes/blob/humble/interbotix_perception_toolbox/interbotix_perception_pipelines/src/pointcloud_pipeline.cpp) for more details.

YumTaha, 2024-11-01:
Thanks, now I can see the CropBox.
I've never had this but any idea why cpu utilization is 100%  when I ran the perception command?

```
ras@pibot:~$ top

top - 11:46:51 up 11 min,  1 user,  load average: 5.30, 3.86, 2.15
Tasks: 282 total,   4 running, 278 sleeping,   0 stopped,   0 zombie
%Cpu(s): 48.9 us,  4.5 sy,  0.0 ni, 43.8 id,  0.3 wa,  0.0 hi,  2.5 si,  0.0 st
MiB Mem :  15847.2 total,   9672.6 free,   2155.0 used,   4019.6 buff/cache
MiB Swap:   2048.0 total,   2048.0 free,      0.0 used.  11623.4 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                                                                                            
   7580 iras      20   0 2552032 180024 117840 S 105.6   1.1   2:10.41 pointcloud_tune                                                                                                                                                                                                    
   7578 iras      20   0  673644  82888  46080 R 104.0   0.5   6:23.34 pointcloud_pipe                                                                                                                                                                                                    
   7576 iras      20   0 1382732 152256  52864 S  67.2   0.9   4:03.15 realsense2_came                                                                                                                                                                                                    
   7594 iras      20   0 3256688 244252 158772 R  63.3   1.5   3:44.95 rviz2                                                                                                                                                                                                              
   1212 iras      20   0 5863968 413892 279484 S  18.9   2.6   2:05.62 gnome-shell                                                                                                                                                                                                        
   3191 iras      20   0 1087112 116584  89456 S  15.8   0.7   4:07.25 Xwayland                                                                                                                                                                                                           
   7584 iras      20   0 1031772 170564 101104 S  11.3   1.1   0:43.52 picture_snapper                                                                                                                                                                                                    
   2163 iras      20   0 3567396 464068 284100 S  10.5   2.9   0:26.18 firefox                                                                                                                                                                                                            
   2689 iras      20   0 2541448 210868 116352 R  10.2   1.3   0:07.12 Isolated Web Co                                                                                                                                                                                                    
   7572 iras      20   0 1089304  27648  19840 S   9.0   0.2   0:35.05 xs_sdk                                                                                                                                            
     26 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/1:0H-events_highpri                                                                                                                                                                                        
     27 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/2                                                                                                                                                                                                            
     28 root     -51   0       0      0      0 S   0.0   0.0   0:00.00 idle_inject/2                                                                                                                                                                                                      
     29 root      rt   0       0      0      0 S   0.0   0.0   0:00.16 migration/2                                                                                                                                                                                                        
     30 root      20   0       0      0      0 S   0.0   0.0   0:00.11 ksoftirqd/2                                                                                                                                                                                                        
     31 root      20   0       0      0      0 I   0.0   0.0   0:00.66 kworker/2:0-i915-unordered                                                                                                                                                                                         
     32 root       0 -20       0      0      0 I   0.0   0.0   0:00.08 kworker/0:1-i915-unordered                                                                                                                                      [... cut ...]
   7586 iras      20   0 1883920 174248 115192 S   0.3   1.1   0:02.82 armtag_tuner_gu                                                                                                                                                                                                    
   8053 iras      20   0   22340   4224   3328 R   0.3   0.0   0:00.12 top                                                                                                                                                                                                                
      1 root      20   0  166824  11468   8140 S   0.0   0.1   0:01.25 systemd                                
```

lukeschmitt-tr, 2024-11-01:
The pointcloud pipeline is very resource intensive when running continuously (like during the tuning process). When actually deployed, the pipeline is run once when its service is called.

YumTaha, 2024-11-01:
Is there a way to minimize it since I am trying to tune the object detection but the pointcloud tuner is barely responding and slow

lukeschmitt-tr, 2024-11-01:
You can try to reduce the fps (or resolution) of the camera's depth module using the [`rs_camera_depth_module_profile`](https://github.com/Interbotix/interbotix_ros_manipulators/blob/humble/interbotix_ros_xsarms/interbotix_xsarm_perception/launch/xsarm_perception.launch.py#L268-L272) launch arg (default 30).

closed 2024-11-06 by YumTaha